micropython/micropython #3114: RFC: Add support for I2C slave mode

mijondano, 2017-05-26:
It seems that some preliminary work has been done in this  [fork](https://github.com/coylen/micropython).

Quoting the author [here](https://hackaday.io/project/12986-nasa-wind-decoder/log/44067-esp8266-i2c-problems):

> Using the existing bit banging I2C master code as inspiration of the structure of C code and the program logic that I had used in my python attempts I was able to implement a very rudimentary routine which provided sufficient functionality for this project.

> Having got this far it is a long term aspiration to develop this code further to provide full I2C Slave functionality and ultimately to pull a full bit banged I2C Driver back to the mainline micropython esp8266 codebase. However, that is a job for another time and another project.

> My micropython fork is at https://github.com/coylen/micropython and **the I2C slave code is located in the drivers folder with corresponding changes to the makefile and modesp.c** to build and expose the commands to the micropython user.

I really believe this would be a very interesting feature for micropython.

peterhinch, 2017-05-27:
I2C is primarily designed for slaves implemented in hardware; the master assumes that the slave is always ready to communicate. The protocol lacks the necessary handshaking for slaves which might not be ready. So if the slave is to do anything aside from servicing the I2C bus the driver needs to be interrupt driven. There is still an issue of speed as you need to guarantee the response time in the presence of other interrupts whose handlers might temporarily disable the I2C interrupt.

So it's rather more difficult than the code referenced above, and I have doubts over whether it could be made to work on platforms with long interrupt latency like the ESP8266/ESP32. But if the problems were solved it would be very useful on Pyboard-like targets.

dpgeorge, 2017-05-28:
I'm keen to (eventually) have support for I2C slave.  But whether it be implemented in software or hardware, the first thing to do is define the API for it.  This has been somewhat discussed before, it would be called machine.I2CSlave and it'd be controlled from Python at a more higher-level than individual I2C transactions.  Eg it would allow you to create either a memory/register based device, or a FIFO.  Feel free to suggest how the API would look and work.

peterhinch, 2017-05-28:
On the KISS principle I'd argue for a UART-like interface compatible with uasyncio.StreamWriter. In other words support only standard bus operations from the master. Do you think a software implementation could work at reasonable speed on the ESP8266?

mijondano, 2017-05-29:
@dpgeorge Actually, as an experiment, a copy pasted the code I mentioned in the OP into machine_i2c and have a working firmware in a esp8266 based board (WeMos D1 mini) with the only method exposed in micropython. If the plan is to have something like machine.I2CMaster and machine.I2CSlave I can help with the job. There is a lot of room for improvement in the current I2C class already, like repeated starts (which implies the possibility of having multi-master setups), and on the slave side implementing clock-stretching should be a must.

@peterhinch Software implementation of master and slave are fully-functional with atmel attinyXX which happen to be much slower clockwise than esp8266. I'm not sure if you refer to overhead caused by micropython itself that could cause reentrancy problems with interrupt based slave implementation. I believe that until it's not tested we won't know for sure.

dpgeorge, 2017-05-29:
> There is a lot of room for improvement in the current I2C class already, like repeated starts

A lot of time/effort was spend on the current I2C (master) class.  It can do repeated starts (see the "stop" parameter to readfrom/writeto).  If there're concrete suggestions, based on real-world usage, for improving machine.I2C then please discuss it in a separate ticket (or ideally on the forum first).

> with the only method exposed in micropython

Which method was exposed and what does it do?

mijondano, 2017-05-29:
Method exposed is **slave_readin(scl, sda, addr_in, nbytes_in)** See below for more details.

[image]

Regarding the rebuke you gave me, it's all my fault to raise an issue here, sorry about that.
I was aware of the stop parameter in writeto/readfrom methods, again that was all my fault because I was thinking really in multi-master setups which is impossible at the moment because with the current implementation a second master trying to take control of the bus will throw an exception.

I only wanted to help here, but that's ok. You can freely close this issue.

peterhinch, 2017-05-29:
I don't think closing the RFC was a good idea.

mijondano, 2017-05-29:
@peterhinch Feel free to correct me if I'm wrong. High hundreds of μs... Let's say a millisecond, for rounding purposes. That means you can't send data faster than at 1000Hz, (or 2000Hz if it were 500μs) over the I2C bus or any other type of communication, spi, serial, onewire, etc, using interrupt driven communication. Even serial at 9600bps would need 1200+Hz.

If it is really that bad, a blocking approach would be a much better but absolutely suboptimal solution.

Is this issue documented anywhere ? I believe it should, mainly to persuade other people going the ISR way.

Regarding the issue itself, I can reopen it if you really see reason.

dpgeorge, 2017-05-30:
> Regarding the rebuke you gave me, it's all my fault to raise an issue here, sorry about that.
> ...
> I only wanted to help here, but that's ok.

Help is very much welcome!  And the best help is precise statements about things that don't work and reasons you ran into such problems.  With precise statements it's much easier to find and fix a problem, or state that it can't/won't be fixed for such-and-such a reason.  In this case there seems to be 2 main things: I2C slave and I2C multi-master capabilities.  Both of these are pretty advanced features but it would be great if MicroPython could (eventually) support them so if you have experience with them then please share your thoughts and experiments about how it could work.

Note that with I2C slave it would also need to be able to write to the bus, not just read.

coylen, 2017-05-30:
HI all,

I am afraid I am responsible for this very ugly code, in my defence I am a structural engineer not a coder.  It was an hack for a specific project to read a marine instrument display repeater output in I2C.  The main issue I found was that i could not get interupts on the esp8266 fast enough to respond to a start condition, so the code when called essentially hangs the esp8266 waiting for a start condition.  This was acceptable in my specific instance, as i knew that the output from the master is on a continuous loop,  but clearly not good enough for mainstream use.  I did manage to get the I2C reading at up to 100khz but the code has been more or less abandoned as due to memory issues and stability I could not get the performance I was after and switched back to a pyboard based solution.

Happy to help in any way to improve and get this working properly, but I doubt my limited skills and understanding would be of any value.

peterhinch, 2017-05-30:
@mijondano Those were my experimental results using MicroPython, and it does set limits to what can be done at that level. I'd be interested to hear from @dpgeorge or @pfalcon whether techniques exist to reduce the ESP latency in C. In other words, what data rate is considered possible for slave mode on the ESP series?

mijondano, 2017-05-30:
@peterhinch Those numbers you get with MicroPython are far, FAR away from what they get with C in [ESP8266 core for Arduino](https://github.com/esp8266/Arduino). Please take a look at [this](https://github.com/esp8266/Arduino/issues/21) an look at the timings for interrupts in post 2. And that person is using a poor-man's signal generator by touching the pin with a ground wire. It's orders of magnitude shorter than your measurements and eventhough they don't seem ultrafast, they look good enough for most interrupt driven scenarios. One question arises: Is this a problem inherent to MicroPython being slow dealing with interrupts or can it be revised and fixed ?

pfalcon, 2017-05-30:
@mijondano, wikipedia to the rescue: https://en.wikipedia.org/wiki/Interpreted_language

peterhinch, 2017-05-30:
@pfalcon There is a big discrepancy between the ESP8266 and the Pyboard in this respect (under MicroPython). The Pyboard's worst-case latency is well over an order of magnitude faster than that of the ESP8266. @robert-hh has reported similar findings. See https://github.com/micropython/micropython/issues/2972 for details of our experiments and those of @dpgeorge.

@mijondano If you look down the results in post 2 you'll find several instances of latencies of multiple ms. (I assume the one of over 4s is some sort of experimental error). The figure of interest (and the one I measured) is the maximum latency rather than the minimum.

In the issue referenced above @dpgeorge lists a number of potential causes of this latency which would seem to be independent of the language in use. This raises a question over the feasibility of implementing fast event driven device drivers on the ESP* platforms in either language.

mijondano, 2017-05-30:
@pfalcon I did know the differences between compiled vs interpreted vs etc, etc. But I've never ever seen an interpreted language being about 2 orders of magnitude slower than it's compiled "counterpart".  Considering that you are one of the fathers of micropython and you are well aware of this issue I can only think that there is no solution neither workaround possible. So I guess that nodeMCU's lua firmware will suffer from this same problem. And Mongoose, ESPBasic,, ESPruino...

@peterhinch Do you think that generating a signal from a wire attached to ground and manually touching a gpio pin is a good source for *max* timings ? The author of the post states:

> The shortest intervals I get are around 12us, which looks sane for the interrupt processing plus Serial.print overhead.

To my understanding he believes that 12μs would be pretty realistic timing.

On the other hand, the ESP8266 seems to have HW spi. There is an I2C  library (maybe more than one) for attinyXX that makes use of SPI HW for I2C communication. But even with this approach, the 2 orders of magnitude when using interrupts won't magically vanish, I'm afraid.

The whole issue is pointless then.

mijondano, 2017-05-30:
Pointless issue. Interrupt driven I2C slave not doable for real world scenario.

closed 2017-05-30 by mijondano

robert-hh, 2017-05-30:
@peterhinch @mijndano Th elatency tests we made were for Python ISR's. An interrupt service implemented by the firmware in C can have much faster response times, espcially if it ensured that it is running from RAM, either by making it's flash page sticky (I do not know if that is possible with the ESP8266) or by copying it to RAM. Being a slave device, it can define it's message structure itself, so buffers can be small.

peterhinch, 2017-05-30:
> An interrupt service implemented by the firmware in C can have much faster response times

Depending on the reason for the high latency. If the underlying RTOS is disabling interrupts for substantial periods all is lost, regardless of language. The measurements implied that instances of the highest latency were infrequent - I don't know if this provides any hint as to the cause. It does suggest a  language independent cause. Aside from the known issue of GC, MicroPython overheads are usually constant.

@mijndano I accept the limitations of using a bit of wire. The parameter of interest is the **worst case** maximum latency - not the best case. The measurements we performed used better methods. They were performed by different people using different techniques. All pointed to occasional instances of high latency even when conditions were set up to minimise this. Under realistic conditions where the WiFi stack is in use, worst case latency may be even higher.

The moot point is whether fast event driven drivers are possible on the ESP* in any language. This has implications beyond I2C slave to proposals for such things as as software UARTs.