micropython/micropython #3115: MAC Address Bug

dbenhart, 2017-05-26:
I believe that there is a bug in the code for ESP8266 Firmware version 1.9 as hosted on the Micropython.org website. The issue has to do with how it prints the MAC address. When I run the command wlan.config('mac') is prints successfully, except for the 4th octet. The correct value for this octet is "5d", but the board PuTTY shows it as "]" which is the ASCII character with the hex code "5d." The board does not seem to have any issues connecting to the wifi network, and an nmap scan of its IP address shows the correct MAC address, so it seems to be limited to the printing of the string.

Other Info:

- firmware version 1.8.6 also exhibits this behavior, but only on this board
- Other boards where the 4th octet is "b6" and "ee" do not exhibit this behavior
    - I have tried all boards using firmware  v1.8.6 and 1.9
- All boards are Huzzah Feathers from Adafruit

If there's any other information that would be helpful in tracking down this bug, please let me know.

dpgeorge, 2017-05-27:
This is just how Python works: bytes objects are printed using ASCII characters where possible (ie if the byte is printable), otherwise it uses the hex notation.  If you want a nice format for printing use `ubinascii.hexlify(wlan.config('mac'), ':')`

closed 2017-05-27 by dpgeorge

dm3942, 2018-12-31:
Another option.

import network
ap_if = network.WLAN(network.AP_IF)
print('starting wifi ap')
ap_if.active(True) # must be true before running config command
mac = ap_if.config('mac')
smac = ""
for ot in list(mac):
    h = hex(ot); smac += h[2] + h[3]
print(smac) # Print mac address without colons 
ap_if.config(essid='SSIDNAME', authmode=network.AUTH_WPA_WPA2_PSK, password="AP_PASSWORD", hidden='False')
print('ap network config:', ap_if.ifconfig())