micropython/micropython #3116: Cannot compile v1.9

gradoj, 2017-05-26:
I'm having trouble compiling v1.9 using cygwin. I don't have any trouble with 1.8.7. Any help appreciated:

`mkdir -p build-PYBV11/usbdev/core/src/
MPY modules/lcd160cr.py
MPY modules/lcd160cr_test.py
Create stmconst build-PYBV11/genhdr/modstm_const.h
Traceback (most recent call last):
  File "modules/lcd160cr.py", line 1
SyntaxError: invalid syntax
Traceback (most recent call last):
  File "modules/lcd160cr_test.py", line 1
SyntaxError: invalid syntax
make: *** [../py/mkrules.mk:117: build-PYBV11/frozen_mpy/lcd160cr.mpy] Error 1
make: *** Waiting for unfinished jobs....
make: *** [../py/mkrules.mk:117: build-PYBV11/frozen_mpy/lcd160cr_test.mpy] Error 1
Create build-PYBV11/pins_PYBV11.c
`
I deleted LCD160cr.py cause I do not need it but I then get:

`CC ../py/../lib/utils/printf.c
Creating build-PYBV11/frozen_mpy.c
CC ../lib/libc/string0.c
CC ../lib/libm/math.c
CC ../lib/libm/thumb_vfp_sqrtf.c
usage: mpy-tool.py [-h] [-d] [-f] [-q QSTR_HEADER]
                   [-mlongint-impl {none,longlong,mpz}] [-mmpz-dig-size N]
                   files [files ...]
mpy-tool.py: error: the following arguments are required: files
make: *** [../py/mkrules.mk:122: build-PYBV11/frozen_mpy.c] Error 2
make: *** Deleting file 'build-PYBV11/frozen_mpy.c'
make: *** Waiting for unfinished jobs....
`

dpgeorge, 2017-05-27:
This is a duplicate of #2961: symlinks don't work on cygwin so you either need to copy the files (in this case lcd160cr.py) or delete them.  After deleting them (everything in stmhal/modules/) try doing "make clean", then it should work.

gradoj, 2017-05-27:
I will have to compile on a linux machine. I get the errors above if I leave or delete the py sym link file and I get this linking problem if I copy the files from drivers into the module directory:
`CC build-PYBV11/frozen_mpy.c
LINK build-PYBV11/firmware.elf
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `arm-none-eabi-ld -nostdlib -L boards -T boards/stm32f405.ld -Map=build-PYBV11/firmware.map --cref --gc-sections -o build-PYBV11/firmware.elf build-PYBV11/py/mpstate.o build-PYBV11/py/nlrx86.o build-PYBV11/py/nlrx64.o build-PYBV11/py/nlrthumb.o....`

dpgeorge, 2017-05-28:
To completely disable the frozen bytecode, try running the following (in stmhal/ dir):
```
$ make FROZEN_MPY_DIR=
```
If it still fails, please run `make V=1 FROZEN_MPY_DIR=` and report the results.

gradoj, 2017-05-28:
Looks like i'm getting the same linker error as above. I also upgraded the compiler to the latest 6.3.1 feb 2017 with no change. Attached verbose output.

[output.txt](https://github.com/micropython/micropython/files/1034224/output.txt)

dpgeorge, 2017-05-29:
The issue is that your libgcc.a exists in directory with "(" and ")" in it, namely `c:/program files (x86)/gnu tools arm embedded/6 2017-q1-update/bin/../lib/gcc/arm-none-eabi/6.3.1/thumb/v7e-m/fpv4-sp/hard/libgcc.a`.  This filename needs to be correctly quoted and/or escaped for it to work.  (This wasn't an issue in v1.8.7 because libgcc.a was not used in that version.)

You can try changing line 64 of stmhal/Makefile to have quotes like this:
```
LIBS = "$(shell $(CC) $(CFLAGS) -print-libgcc-file-name)"
```
Otherwise, if this doesn't work, try changing this line to simply `LIBS = `.  You don't really need the library.

gradoj, 2017-05-29:
The quotes worked for me. Thanks.

LIBS = "$(shell $(CC) $(CFLAGS) -print-libgcc-file-name)"

closed 2017-05-29 by gradoj

hhycon, 2017-06-07:
Just want to add, if you enable symlinks `-c core.symlinks=true` while cloning the repo I'm able to successfully compile.

commit da71f55e (2020-06-02): stm32/Makefile: Quote libgcc path so spaces are not an issue.

Fixes #3116.