micropython/micropython #3119: machine.Pin Firmware V1.9 breaks code on some platforms

peterhinch, 2017-05-29:
See [this thread](https://forum.micropython.org/viewtopic.php?f=16&t=3354&p=19929#p19929).
tl;dr summary:
The pin.high() and pin.low() methods are deprecated but are used in existing code. On the Pyboard they still work, but on ESP8266 an Attribute Error is raised.

It was pointed out that this makes it hard to write code compatible with old and new firmware.

Is this behaviour by design?

pfalcon, 2017-05-29:
There never have been documented machine.Pin.high() and machine.Pin.low() methods. Code which uses them is erroneous and needs fixing.

dpgeorge, 2017-05-30:
See #3025 for previous discussion.

> The pin.high() and pin.low() methods are deprecated but are used in existing code.

There shouldn't be any use of high/low in this repo anymore.  Only pyboard's pyb.Pin implements them.

> It was pointed out that this makes it hard to write code compatible with old and new firmware.

Old and new firmware both support `pin.value([x])` and the short-hand `pin([x])`, so they can be used to write compatible code.

mcauser, 2017-05-30:
ESP8266 Quick Ref mentions Machine.Pin `.high()` and `.low()`:
http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html#pins-and-gpio

The detailed Machine.Pin page does not:
http://docs.micropython.org/en/latest/esp8266/library/machine.Pin.html

pfalcon, 2017-05-30:
> ESP8266 Quick Ref mentions Machine.Pin .high() and .low(): http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html#pins-and-gpio

Fixed. That's exactly the things we're trying to resolve - instead of many things which don't work, we'll have few things which work consistently.

pfalcon, 2017-05-30:
docs.micropython.org will be updated tomorrow.

scargill, 2017-05-30:
Hi there..

A comment was made that there has never been documented pin.high() and pin.low() methods.

I hope this then is helpful.

Yesterday as a complete microPython newbie - I downloaded the 1.9 code (latest) and went to the ESP8266-specific documentation - 1.9 latest.

https://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/repl.html

See section 2.3 therein - the very first example of port usage - sorry - but it absolutely does tell you to use high() and low().  I don't know if this is "official" documentation but it looks official.

robert-hh, 2017-05-30:
> Old and new firmware both support pin.value([x]) and the short-hand pin([x]), so they can be used to write compatible code.

So if in expressing the intention in the code the level of the output is of importance, pni.value(x= or pin(x) should be used. If not, pin.on() and pin.off() may be used, especially if depending on the circuitry and as told in the documentation the opposite may happen. And besides that, I do not know what the ON-state of a pin is. The best I could imagine os low vs. high impedance equivalent to a switch being closed or open. For the signal class, this is pretty clear.

pfalcon, 2017-05-30:
Documentation at http://docs.micropython.org/en/latest/esp8266/ was updated.

scargill, 2017-05-30:
Excellent!

scargill, 2017-05-30:
This is great stuff – I’m having fun picking this up on my little ESP board.

 

Pete.

peterhinch, 2017-05-31:
The documentation issue is fixed so I'll close this.

closed 2017-05-31 by peterhinch