micropython/micropython #3120: NUCLEO_F767ZI DAC compile error

mikeredbike, 2017-05-29:
Using Vsn 1.90 I turned on the MICROPY_HW_ENABLE_DAC flag in mpconfigboard.h then ran  make BOARD=NUCLEO_F767ZI V=1.  Then got this compile error.  Any advice?

CC dac.c
arm-none-eabi-gcc -I. -I.. -Ibuild-NUCLEO_F767ZI -I../lib/cmsis/inc -Icmsis/ -Ihal/f7/inc -Iusbdev/core/inc -Iusbdev/class/inc -Wall -Wpointer-arith -Werror -std=gnu99 -nostdlib -DFFCONF_H=\"lib/oofatfs/ffconf.h\" -DSTM32F767xx -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fsingle-precision-constant -Wdouble-promotion -mtune=cortex-m7 -mcpu=cortex-m7 -DMCU_SERIES_F7 -Os -DNDEBUG -Iboards/NUCLEO_F767ZI -DSTM32_HAL_H='<stm32f7xx_hal.h>' -fdata-sections -ffunction-sections -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -c -MD -o build-NUCLEO_F767ZI/dac.o dac.c
dac.c:72:8: error: unknown type name 'DAC_HandleTypeDef'
 STATIC DAC_HandleTypeDef DAC_Handle;
        ^
dac.c: In function 'dac_init':

dhylands, 2017-05-29:
You also need to enable the DAC in the HAL configuration file:
https://github.com/micropython/micropython/blob/d5b8825d5f5c6110f9cf3bd39848cfe94137d291/stmhal/boards/NUCLEO_F767ZI/stm32f7xx_hal_conf.h#L61

dpgeorge, 2017-05-30:
DAC has not been tested on the F7 MCUs.  If you can get it working please let us know and it can be enabled by default.

mikeredbike, 2017-05-30:
As suggested by dhylands I defined HAL_DAC_MODULE_ENABLED in stm32f7xx_hal_conf.h and it compiled fine.  DAC outputs PA4 and PA5 are generating triangle wave forms.  With the hal_conf change you  can enable DAC by default. Thanks.

dpgeorge, 2017-05-30:
@mikeredbike thanks for the info.  DAC for this board has been enabled in 821dc27eecf2d47d175222e9259d5b287b45380c

closed 2017-05-30 by dpgeorge