micropython/micropython #3121: OSError 28 when trying to write file

craftyguy, 2017-05-30:
I receive an error indicating that there is no free space when trying to write a very simple file on the ESP8266 port:

MicroPython v1.9-4-gd5b8825d-dirty on 2017-05-29; ESP module with ESP8266
Type "help()" for more information.
>>> f = open('test', 'w')
>>> f.write('hi')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 28
>>> os.statvfs('flash')
(4096, 4096, 860, 0, 0, 0, 0, 0, 0, 255)
>>> os.listdir()
['boot.py', 'main.py']

robert-hh, 2017-05-31:
Hi @craftguy, Try erasing the flash with esptool.py and reloading the image. The layout of the flash has been changed some version ticks below the road.

craftyguy, 2017-05-31:
@robert-hh thank you so much for the suggestion, this seems to have resolved the issue for me! I'm now sorting through some weird instances of modules created on the esp (e.g. f.write(..)) not being able to import, but copying them over webrepl works. Will file a separate ticket if I can disposition it to something other than a mistake on my part!

closed 2017-05-31 by craftyguy

curlyz, 2020-03-12:
Maybe this will help.
I have encounter this problem, try to split the file content into small chunk ( 256 bytes ) and then do f.write , maybe this has something to do with flash fragmentation ?

tve, 2020-03-12:
OSError is "No space left on device", so fragmentation may well play a role.