micropython/micropython #3123: Windows 10 Creator Breaks uPY Flash Drive

kwagyeman, 2017-05-31:
It seems the windows 10 creator update breaks the uPY flash drives somewhat. For example, on all OpenMV Cams the uPY drive now appears about 5 minutes after being plugged into the computer.

See here:

https://forum.micropython.org/viewtopic.php?f=2&t=3369&p=19971#p19971
https://github.com/openmv/openmv/issues/236

Any thoughts? You can see the uPY drive as a windows device more or less immediately in the windows device manager. But, the USB disk just won't appear for a while.

dpgeorge, 2017-06-01:
USB MSC is a mature and often used protocol across all OS platform, so this is a strange thing to happen.

Obvious tests to do are:
- try a USB flash drive, with and without partitions; does it work?  If so it would be good to see a log of the USB transactions and a dump of the USB flash blocks.
- try other dev boards, eg an mbed-enabled board like the microbit: does it's MSC device show up straightaway and work as expected?

kwagyeman, 2017-06-01:
I have a USB analyzer. I'll provide some logs in a bit. I also bought a pyboard to see the issue.

kwagyeman, 2017-06-04:
The commit you did works.

robert-hh, 2017-06-05:
Please re-open it until the PR of  @dpgeorge  is merged. He can close it then.

dpgeorge, 2017-06-08:
I managed to verify this problem and check that the fix does indeed fix it.  Now fixed in the master branch.

closed 2017-06-08 by dpgeorge