micropython/micropython #3124: Unable to create builtin object from subclassed object

MrSurly, 2017-05-31:
Using the baseline https://github.com/micropython/micropython-esp32/tree/esp32

Best explained with code:

```python
class Y(dict): 
    pass
y = Y()
y[1] = 'one'
print(y)
dict(y)
```

Expected output (per CPython): 

```python
{1: 'one'}
{1: 'one'}
```

Actual output:

```python
{1: 'one'}
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'int' object is not iterable
```

I dug into this a bit:

In `py/objdict.c`, `dict_make_new` calls `dict_update`, which incorrectly identifies the type of the parameter as being something other than a `dict`.  

How bound `dict` methods work for object `y`  (e.g. `y.items()`) is above my paygrade, but they do work. Nonetheless, I tried modifying `MP_OBJ_IS_DICT_TYPE` to call `mp_obj_is_subclass_fast`.  This doesn't work -- it actually gets a different `mp_obj_t` pointer than what is passed to `y.items()`.  While the error goes away, the dict returned from the `dict(y)` line above is empty, since it's looking for `dict` items in a `class Y` object.

dpgeorge, 2017-06-01:
Duplicate of #2749.

dpgeorge, 2017-06-01:
@MrSurly : if you like, please comment on the above-linked issue about your particular use/need for this feature.

closed 2017-06-01 by dpgeorge