micropython/micropython #3125: Missing documentation in esp8266 about vfs

Cediddi, 2017-06-02:
Hi,
With 1.9 you've released a huge thing for the community, new filesystem api. When I tried to read the documents I've failed to find os.mount function. I'd like to help you about the documentation of that section and the old code examples on sdcard driver. Where should I start for the api documentation? I'm a bit confused with the new api overall.

Thanks for the great feature, I'll be writing about it in a local maker magazine Çırak.

dpgeorge, 2017-06-07:
If you'd like to contribute docs for the mount/umount functions then they would go in docs/library/uos.rst.

deshipu, 2017-07-25:
I wonder how this process of contributing documentation for code written by someone else is supposed to work? Are we supposed to read the code, reverse-engineer it, figure out what the developer could have in mind, and write that down? How can we know what parts of the code are "official" APIs and what are just implementation details, omissions or even bugs? Once we write this documentation, what guarantee do we have that it won't suddenly change in the next version, with the "explanation" that "it was never officially documented", like the ``pin.high()`` and ``pin.low()`` did?

Why can't the developer who comes up with the API design in the first place document it, and change that documentation when he changes the API? It seems to me that this would work much better both in terms of accuracy and in terms of committing to a certain API (and having to do extra work to change it, which would hopefully result in fewer such changes).

WayneKeenan, 2017-07-26:
Talk/email/irc the developer(s) and ask those kinds of questions.

dpgeorge, 2018-03-07:
In 8359210e71ee0c46434ef8b1056627597f0bf0bf documentation was added for `uos.mount()`, `uos.umount()`, `uos.VfsFat()` and a description of block devices.

closed 2018-03-07 by dpgeorge