micropython/micropython #3129: btree usage

imihajlow, 2017-06-05:
Dear all,
I'm trying to use btree module in my project. First of all, in the [example](https://docs.micropython.org/en/latest/esp8266/library/btree.html) 'w+b' file mode is used, which means file is truncated on every open. I want my database to be persistent, so I use mode 'a+b'. First open works fine:
```
f = open('test.db', 'a+b')
b = btree.open(f)
b[b'a'] = b'123'
b.get(b'a') # prints b'123'
b.close()
f.close()
```
Then, after I open the file for the second time, OSErrors show up:
```
f = open('test.db', 'a+b')
b = btree.open(f) # OSError: 79
```
If I try to seek the file to its end, OSError pops up in another place:
```
f = open('test.db', 'a+b')
f.seek(0, 2)
b = btree.open(f) # OK
b.get(b'a') # OSError: [Errno 22] EINVAL
```

Is this me who is wrong or btree is broken?

pfalcon, 2017-06-05:
You. See https://github.com/micropython/micropython/issues/3086

dpgeorge, 2017-06-08:
@pfalcon is there any (documented) work around for this?

pfalcon, 2017-06-08:
@dpgeorge: Workaround for what? Functionality in #3086? Well, the usual one a Python programmer would do: https://github.com/pfalcon/micropython-btreedb/blob/master/btreedb.py#L16

peterhinch, 2017-06-08:
CPython supports read/write random access files via [mmap](https://docs.python.org/3/library/mmap.html).

dpgeorge, 2017-06-11:
> unctionality in #3086? Well, the usual one a Python programmer would do: https://github.com/pfalcon/micropython-btreedb/blob/master/btreedb.py#L16

Yes, it might be nice for to add this 4-liner to the btree example in docs/library/btree.rst, since it's a very common thing to need to do to use the btree module in a real application.

peterhinch, 2017-06-11:
In many applications you want to open an existing file read-write. Is there actually a way to use btree with a persistent writeable file?

pfalcon, 2017-06-11:
Yes. If there weren't such a way, what would be the purpose of btree module?

pfalcon, 2017-06-11:
> Yes, it might be nice for to add this 4-liner to the btree example in docs/library/btree.rst

Done in 6ca086a89a86ab9b01ad8e1ec4621996b08f06e0

closed 2017-06-11 by pfalcon