micropython/micropython #3132: USB device product description is not working on windows

Arrowana, 2017-06-05:
The USB product description doesn't show up in windows 10.
It is currently hard to build automation because there is no way to recognise a specific pyboard without engaging any serial communication.

The functionality is present in usbd_desc.c however for some reason it does not work on windows.

The following screenshot shows that an Arduino can transmit his product description properly while the STM running micropython can't.

COM31 is a STM32F439 running micropython
[image]

dpgeorge, 2017-06-07:
This is most like because the VID/PID are not registered with the USB group, and so Windows doesn't have the VID/PID in it's database.  If you are building a product and want to have your device show up with its name then you'll need to purchase a VID and use that one.

stinos, 2017-06-07:
I think http://pid.codes/ still has codes left, maybe get one there?

jonnor, 2024-07-19:
MicroPython now runs on many ST devices, not just pyboard. Proposing to close this as not planned.

closed 2024-07-19 by robert-hh