micropython/micropython #3136: Prepare for release v1.9.1

dpgeorge, 2017-06-08:
The USB SCSI bug discussed in #3132 and fixed by #3126 is quite serious for Windows 10 users, so it would be good to get a version out soon that has this fix in it.

There have also been some other minor fixes of regressions introduced by the new VFS.

Timescale for v1.9.1 would be end of Sunday 11th June 2017, unless there are concerns that that is too soon.

pfalcon, 2017-06-08:
Well, if you feel like doing it that way, sounds good, just need to make sure we don't commit bigger feature/refactors until then. Otherwise, 1.9 had not-really-working uasyncio for esp8266, with few fixed and optimizations already applied, so releasing that soon only sounds good, just need to test it more (but unfortunately I'm a bit short on time/possibilities to do that as thorough as I'd want).

dpgeorge, 2017-06-11:
Ok, uasyncio examples now work on esp8266.

dpgeorge, 2017-06-11:
v1.9.1 was tagged in 869cdcfdfc860b1177baf9b0f8818915ba54f3f5.  esp-extra-scripts was rebased on it and force-pushed.

closed 2017-06-11 by dpgeorge