micropython/micropython #3138: esp8266 modules/_boot.py gc threshold calculation

peterhinch, 2017-06-11:
This has
```python
gc.threshold((gc.mem_free() + gc.mem_alloc()) // 4)
```
which sets the threshold to 1/4 of the total RAM regardless of the amount allocated. Shouldn't this read
```python
gc.threshold(gc.mem_free() // 4 + gc.mem_alloc())
```
to account for RAM already allocated?

pfalcon, 2017-06-11:
The current code is implemented as was intended. If you have done extensive research proving that the other formula leads to sustainably better results, feel free to share it.

peterhinch, 2017-06-11:
My extensive research comes from reading [the docs](https://docs.micropython.org/en/latest/pyboard/reference/constrained.html?highlight=threshold).

dpgeorge, 2017-06-11:
See the discussion about changing this factor from 1/2 to 1/4: https://github.com/micropython/micropython/commit/c141584e1e30b6dd2268d8831b706dcb9ab25c7c

See also the commit message that changed to 1/4: 0d221775f5ab458358ee873b70ad16f284a9b1cb

peterhinch, 2017-06-11:
It's the position of the brackets rather than the factor which concerns me. Which is correct, the code or the docs? I've used that from the docs extensively in my projects. It would be good to know if it's incorrect, along with the reason why.

peterhinch, 2017-06-22:
Please could you confirm if this understanding is correct. The aim of the formula in _boot.py is to control fragmentation in the period while modules are being loaded. When the user application is running, the formula in the docs should be applied.

pfalcon, 2017-06-24:
Let's keep this open to fix possible issue in the docs (when someone will give it a sober look).

> should be applied

There're no "should's". It's *heuristic*, so it may be "more logical" and "less logical" (just like you spotted). The results don't directly depend on rational hunches though due to butterfly effect.

pfalcon, 2017-06-24:
As a first step, gc.threshold() is documented in c4e3a03fa53bf456e13f930bc6187f203ca77e71

pfalcon, 2017-07-01:
Ok, while converting markup of this doc (e334b6b6d21ee0d92a4c91812f7ede05d1d0b447), I also look into more detail into this issue.

First of all, the discussed formula is mentioned twice in the doc:

1. First time, it's introduced by words "The discourse on this is somewhat involved. For a ‘quick fix’ issue the following periodically:". Suggestion: remove the paragraph with formula completely. This should be an introductory doc, it indeed can't and shouldn't go into involved matters, and it it doesn't, it shouldn't spit magic formulas either. That can only lead to confusion.

2. 2nd time, it's described as "This threshold can be adapted as execution progresses: (formula). This will provoke a GC when more than 25% of the currently free heap becomes occupied.". The 2nd clause of the description is incorrect, that formula won't provoke a GC when more than 25% of the currently free heap becomes occupied (see docs above on how gc.threshold() works). Secondly, "This threshold can be adapted as execution progresses" can be extended by saying "Also, threshold can be adapted depending on the phase of the moon". Doing so would be as useful as adapting it as execution progresses - unless proven that doing so gives any benefit. The doc doesn't prove it, nor it's possible to prove. Then, the doc, being introductory, should show a user the *easiest to understand* formula, and explain how it works. From that, anyone with inclination to think, will figure out that threshold can be updated in any way at all. Suggestion: replace the formula with gc.threshold((gc.mem_free() + gc.mem_alloc()) // 2) and explain how GC works with it.

@peterhinch : Let me know if you disagree, otherwise I'll prepare a patch.

peterhinch, 2017-07-01:
Sounds good to me.

dpgeorge, 2017-07-02:
> Suggestion: replace the formula with gc.threshold((gc.mem_free() + gc.mem_alloc()) // 2) and explain how GC works with it.

Did you mean `// 4`?  That would then match with how esp8266 does it.