micropython/micropython #3139: esp8266 makefile Add support for per-project frozen module directory

peterhinch, 2017-06-11:
Given the limited space for frozen bytecode this is useful. The following seems to work. Suggested comments included.
```
# To use frozen bytecode, put your .py files in a subdirectory (eg frozen/) and
# then invoke make with FROZEN_MPY_DIR=frozen (be sure to build from scratch).
# The following files must be included or symlinked:
# _boot.py flashbdev.py inisetup.py
FROZEN_MPY_DIR ?= modules
```

pfalcon, 2017-06-11:
So, what's reported/proposed here?

peterhinch, 2017-06-11:
On the Pyboard we can override the default frozen modules directory so each project can have a different set of frozen modules. I suggest implementing this useful feature on the ESP8266. In my testing this one-liner does the trick.

pfalcon, 2017-06-11:
What makes you think it's not possible to override it for ESP8266?

peterhinch, 2017-06-11:
On the Pyboard I export FROZEN_MPY_DIR prior to running my build script. On the ESP8266 that didn't work until I modified the makefile to match the stmhal makefile. If this is the wrong approach, please advise.

pfalcon, 2017-06-11:
That's not what you wrote in the original comment:

> then invoke make with FROZEN_MPY_DIR=frozen

If something doesn't work, use what works. If in doubt, make manual has all the answers (if not, report problem there). Avoid changing the world (it's too big), adapt yourself instead.

peterhinch, 2017-06-11:
OK, I'll adapt my scripts. It seemed any easy way to make it behave like stmhal, accepting command line args or environment variables.

dpgeorge, 2017-06-13:
stmhal's Makefile uses `?=` to set the FROZEN_MPY_DIR, unlike esp8266 which uses just `=`.  esp8266 uses `?=` in other places for things that are sensible to override (eg PORT, BAUD) so it makes sense to do the same thing for the frozen dir config.

pfalcon, 2017-06-13:
Or maybe it makes sense to set stmhal to `=` because arguably, it's never sensible to pass any adhoc settings (outside those defined in a common standard like POSIX) via environment. The only case when it's sensible is if you use one board ever and forever. If you use two, one of them once a year, you're already guaranteed to have problems, twice: once when you connect the rare board and wonder why it doesn't work, then remember (after a year) that you need to set env vars for it. Then you switch to the usual board and wonder it doesn't work, while it did for a year, finally remembering that you overrode env for another board.

Generally, there're millions of Makefiles in the world with few to dozen `=` variable assignments in them, such folks are going to patch tens millions of lines, or what?

peterhinch, 2017-06-13:
Only a problem if you set the envars globally. So export them from a calling script.

pfalcon, 2017-06-13:
Exporting vars from a calling script will help to patch millions of projects with useless changes, did I understand that correct?

dpgeorge, 2017-06-14:
> Or maybe it makes sense to set stmhal to = because arguably, it's never sensible to pass any adhoc settings (outside those defined in a common standard like POSIX) via environment.

This would break my standard workflow, which is to create a custom "makefile" that sets the variables I'm interested in and then includes the main Makefile.  I don't need to remember anything, just that each makefile has all the relevant settings (and in some cases multiple sets of settings in one makefile).

pfalcon, 2017-06-14:
Ok, so let's do this change. My point is, after that, esp8266/Makefile will still contain 25 variables set by `=`, then there're such vars in py/*.mk, then there're growing number of other ports. So there, never would be shortage of reports "change = to ?= for me".

dpgeorge, 2017-06-15:
Done in 76ec04a6d93668806409b6f5b2788a4043fb7bb5

closed 2017-06-15 by dpgeorge