micropython/micropython #3140: Upgrade axTLS to 2.1.3

pfalcon, 2017-06-11:
I've rebased MicroPython axTLS onto 2.1.3 (the latest). The rebasing was pretty clean. 2.1.3 adds support for TLS 1.2 (some sites effectively deprecate/disable anything below) and SNI, both must features. Basic testing on x86 shows that it works. But upgrade increases codesize (x86) by 12K (which is very much by both absolute and relative terms, the latter is by third).

I guess we should upgrade nonetheless, as lack of the features above highly limits TLS support.

I'll try to look at what may have caused code increase (thought that's when it turns from pretty clean upgrade into a time sink).

pfalcon, 2017-06-11:
Weird, summarized increase of code size of libxtls.a between new and old version is 36654 - 34643 = 2011 . And yet binary increase is:

~~~
 381466	   4652	   1416	 387534	  5e9ce	micropython.axtls_new_no_sni
 368856	   4648	   1416	 374920	  5b888	micropython.axtls_old
~~~

pfalcon, 2017-06-11:
Ok, got it - TLS 1.2 requires SHA256, and gigantic implementation gets pulled in:

~~~
10501	sha256.o (ex libaxtls.a)
~~~

pfalcon, 2017-06-11:
Ok:
~~~
 371106	   4652	   1416	 377174	  5c156	micropython
~~~

dpgeorge, 2017-06-11:
What is the outcome, were you able to use the existing (minimal) sha256 implementation in extmod/crypto-algorithms instead of the one shipped with axtls?

pfalcon, 2017-06-11:
Yup, https://github.com/micropython/micropython/issues/3140#issuecomment-307623253 gives the result. "Everything" works.

pfalcon, 2017-06-11:
Waiting for https://github.com/micropython/micropython/issues/3136 to start pushing it.

pfalcon, 2017-06-12:
Update: I see apparent memory corruption with new version, will be investigating. In the meantime, axTLS 2.1.3 + uPy patchset is at https://github.com/pfalcon/axtls/tree/micropython-2.1.3

pfalcon, 2017-06-13:
Ok, memory corruption was due to over-optimization on my side which opened Pandora box of issues, so was reverted. Upgrade happened in f01c1c6b35a94d717c2174662fb242bf235f6978 and 75c3f2a7ab508f2067184646989d88414e1f0fea. Note that upgrades brings (via uPy patchset on top of axTLS) debug dumping in case of TLS buffer overflow (which is a common issue on esp8266): https://github.com/pfalcon/axtls/commit/9b3092eb3b4b230a63c0c389bfbd3c55682c620f

pfalcon, 2017-06-13:
server_hostname parameter (SNI) added in 82b9915b34fb3973ca7657dd8ef4f3ad3681f161

pfalcon, 2017-06-13:
In a47b8711316a4901bc81e1c46ce50de00207c47f, TLS record buffer size is bumped to 5K.

pfalcon, 2017-06-13:
Announced on the forum: https://forum.micropython.org/viewtopic.php?f=16&t=3469

pfalcon, 2017-06-14:
micropython-lib modules updated to support SNI.

Closing.

closed 2017-06-14 by pfalcon

mleonard87, 2017-06-19:
This is great! Will this be on the esp8266 build too? When will it go into a release?

pfalcon, 2017-06-19:
esp8266 port was the primary reason of this upgrade. It's available in master, i.e. in daily builds, and will be in the next release (1.9.2, ETA is unknown so far).

kannan95, 2017-07-06:
@pfalcon When will be the 1.9.2 version release? I want the TLS 1.2 support in micropython. If the release is not public yet, can i build my own to support TLS1.2?. Since i'm getting the ssl_handshake_status: -40 when i send a post request of a https url.

dpgeorge, 2017-07-06:
@kannan95 as said above, you can download the "daily builds" from here: https://micropython.org/download/#esp8266 .  If you want to include all the extra scripts/modules that come in a stable release then you'll need to build it yourself (which is not that difficult).

kannan95, 2017-07-06:
@dpgeorge the daily builds are still v1.9.1. In which build the ssl_handshake_staus:-40 problem got closed? All i wanted for now is sending post request to a https url, where im getting ssl_handshake_status:-40 error. If there is any daily build for this problem, please link the correct build i want to download. If there is no build for the error, then guide me how to build own ^^

pfalcon, 2017-07-06:
@kannan95 : To clarify again, you will need to try the daily build, then if you experience issues, submit a separate ticket based on the suggestions in https://www.chiark.greenend.org.uk/~sgtatham/bugs.html .

kannan95, 2017-07-06:
Sure! Thanks! @pfalcon

coolaj86, 2018-12-01:
I've been trying to figure out how to choose the SNI and my google searches are leading me to these bug reports, but not to documentation.

This is the documentation that I'm finding:
https://docs.micropython.org/en/latest/library/ussl.html

Can someone point me to docs or an example showing how to use Servername Indication?

Many thanks.

dpgeorge, 2018-12-02:
> Can someone point me to docs or an example showing how to use Servername Indication?

It looks like it's not documented... to use it pass the `server_hostname` keyword argument to `ussl.wrap_socket()`.  See eg `tests/net_inet/test_tls_sites.py` and `tools/upip.py`.