micropython/micropython #3141: RFC: Support for assigning to module variables of C native modules

pfalcon, 2017-06-11:
We have at least couple of usecases where we need to handle assignment to variables in C modules:

* Overriding sys.stdin/sys.stdout
* Assigning to time.timezone, per https://github.com/micropython/micropython/issues/2130

Proposed way to handle that:

We already have `__init__` method for initialization of native modules. We could have `__setattr__` which would be called on attempt to assign to a var in a module. This assume that each var assigned to is "real", i.e. present in a module dictionary, so only writes needs to be "virtualized".

dpgeorge, 2017-06-13:
> We could have __setattr__ which would be called on attempt to assign to a var in a module.

Where does the assigned value get stored, does the module have local state?

> This assume that each var assigned to is "real", i.e. present in a module dictionary, so only writes needs to be "virtualized".

How does the read work to get back the assigned value?

pfalcon, 2017-06-13:
> How does the read work to get back the assigned value?

As mentioned above, the module-level variables are present in the module dictionaries, and read from there as usual (but apparently are back by RAM-based storage instead of ROM).

 > Where does the assigned value get stored, does the module have local state?

Given the "read" requirement above, it would be definitely stored as normal mp_obj_t in a module dict. But `__setattr__` can validate that value is correct (to not validate on each use). Another thing `__setattr__` can do is to cache (processed) value elsewhere. E.g., it can cache read/write methods of a stream. (This usecase comes from overriding sys.stdin/stdout).

Disclaimer: I had this idea in mind for a long time (year+), coming from an adhoc patch I made to allow to override sys.stdin/stdout. There might have been reason I didn't submit this RFC before, so there may be some "white spots" in this proposal. (One is GC handling for values of such vars.)

dpgeorge, 2017-06-14:
> and read from there as usual (but apparently are back by RAM-based storage instead of ROM).

So you suggest that modules like sys will convert their entire globals dict/table to RAM?  That suddenly increases RAM usage by the size of the entire table.

pfalcon, 2017-06-14:
In the case of sys, only stdin and stdout entries in the module dict, #ifdef'ed of course.

dpgeorge, 2022-03-30:
Support for mutable attributes in C modules was added in 3356b5ef8d1b232a1df493f70f0ba434f293fe11

closed 2022-03-30 by dpgeorge