micropython/micropython #3149: SNI not working on unix

annejan, 2017-06-15:
I have changed my `upip` variant to take into account `server_hostname` for SNI.
`s = ussl.wrap_socket(s, server_hostname=host)`

This works perfectly on the ESP32, however on the unix variant I get `ssl_handshake_status: -40`

I have tried cleaning `actls` and even a brand new checkout . .

dpgeorge, 2017-06-15:
> This works perfectly on the ESP32

Note that ESP32 uses mbedtls, not axtls.

annejan, 2017-06-15:
Correct . . hence me thinking it has something to do with axtls . .

The server I'm trying to connect to is badge.sha2017.org it is behind a loadbalancer and the non-SNI fallback cert is different from the requested (*.sha2017.org) wildcard . .

pfalcon, 2017-06-15:
@annejan: Unless you have proofs that SNI is not working, it would be fair to name ticket "ussl on unix: ssl_handshake_status: -40, suspected SNI not working".

annejan, 2017-06-15:
Working on those proofs . . 
It's unfortunately kind of hard to check deeper down currently (lots of reading up to do) . .

pfalcon, 2017-06-15:
See https://github.com/micropython/micropython/pull/3150

pfalcon, 2017-06-15:
> It's unfortunately kind of hard to check deeper down currently (lots of reading up to do) . .

Ack, and it's the same for everyone.

annejan, 2017-06-15:
After attempting a long list of SNI and other domains with test script from https://github.com/micropython/micropython/pull/3150 I have come to the conclusion it's not SNI and probably something todo with the load balancer of the domain.

Closing issue. . sorry for the inconvenience!

annejan, 2017-06-15:
Found the issue to be that `axtls` only supports TLS_RSA
[image] server only supports DHE and ECDHE

annejan, 2017-06-15:
Unfortunately this is an issue with axtls . .
http://axtls.sourceforge.net/specifications.htm
Unsupported: Anonymous or Ephemeral DH.

closed 2017-06-15 by annejan