micropython/micropython #3151: Possible to use mbedtls in unix port.

annejan, 2017-06-15:
Since I found out #3149 axtls only supports TLS_RSA_WITH_AES_* and none of the more modern DHE/ECDHE handshakes I was wondering if it is easy to set-up the unix port of micropython to use mbedtls instead?

Googling around didn't answer this question for me.

pfalcon, 2017-06-15:
"Is it easy?" is a strange question. Asked somewhere on stackoverflow, it would be closed by its ~~fascistic~~ moderators as highly subjective or something. Not going to close here, but what answer do you really expect except highly subjective? I did it, and remember nothing about it. Given that I don't consider myself to posses more "fu" than other people, then it must have been easy ;-).

Perhaps, you should try it yourself, and tell google how it was, so it knew that afterwards ;-).

annejan, 2017-06-15:
Well .. it seems it was just change the `unix/mpconfigport.mk`
And adding `../extmod/modussl_mbedtls.c \` to the `unix/Makefile`
But it does seem to have quite a bit of side-effects still . .

pfalcon, 2017-06-15:
> And adding ../extmod/modussl_mbedtls.c \ to the unix/Makefile

Unlikely given https://github.com/micropython/micropython/blob/master/py/py.mk#L228

What you're expected to care about is mbedTLS config, which isn't supplied by uPy unlike for the official integration with (patched!) axTLS.

annejan, 2017-06-15:
You were right, and indeed the config is not what it should be on some systems . .
Thanks sofar . . I'm closing this issue . .
There will be a writeup of how to use mbedtls on the unix for on our project page https://wiki.sha2017.org/w/Projects:Badge/Emulator

closed 2017-06-15 by annejan

manningt, 2018-12-09:
Hello @annejan 
I tried the changes required to enable mbedtls on macOS 10.14 as performed in the Badge project, but used the latest version of micropython instead of the Badge fork of micropython-esp32.

Here were the steps:
```
1) download and build mbedtls in the micropython/lib directory:
-- git clone https://github.com/ARMmbed/mbedtls.git
-- cd mbedtls
-- make
2) change mpconfigport.mk:
-- MICROPY_SSL_MBEDTLS = 1
-- MICROPY_SSL_AXTLS = 0
3) run make in the ports/unix directory
```
Everything compiles, but I get the following link error:
```
Undefined symbols for architecture x86_64:
  "_mp_module_ussl", referenced from:
      _mp_builtin_module_table in objmodule.o
ld: symbol(s) not found for architecture x86_64
```
One possibility is that macOS CLANG target does not match:
```
clang -v
Apple LLVM version 10.0.0 (clang-1000.11.45.5)
Target: x86_64-apple-darwin18.2.0
```
Do you have suggestions for how to correct the linker error?   Thanks in advance.

m0ot, 2019-03-29:
Hi,
I'm new with programming libraries and I'm looking for help. I've tested MicroPython on esp32 using MQTT libarary secured by ssl and it worked like a charm. Now, MicroPython supports axTLS by default. axTLS supports until TLS 1.2 only. However, I want to create an MQTT connection secured by TLS 1.3 which already supported by mbedTLS. As I understood, it's possible in MicoPython to link mbedTLS instead of axTLS.

My questions:
Which files in MicroPython must be modified in order to link mbedTLS correctly? and
which tools do I need for developing and testing my work?
thanks
Ahmed

manningt, 2019-03-30:
unix/makefile: add mbedtls as a deplib: deplibs: libffi axtls mbedtls
unix/mpconfigport.mk: change MICROPY_SSL_AXTLS to 0 and MICROPY_SSL_MBEDTLS to 1.

This worked on MacOS when I built it 4 months ago, but I haven't built it recently.

m0ot, 2019-03-31:
but these modifications are to link mbedTLS in Unix port. But I would like to link mbedTLS in ESP32 port and then test the modified code on the ESP32 board. Is it possible?

manningt, 2019-03-31:
I believe the ESP32 port uses mbedtls, not the axTLS.  Check out the ESP32 Makefile.

m0ot, 2019-04-02:
Yes you're right. However, I just tried it again. I implemented the latest esp32 bin file https://micropython.org/download#esp32 on the board. Then I tried to speak secured mqtt connection using the ussl.wrap_socket function https://docs.micropython.org/en/latest/library/ussl.html but it speaks only TLS 1.2
I'm a bit confused! If esp32 bin file supports the open source mbedtls then it should speak with the server TLS 1.3. Any ideas?

manningt, 2019-04-02:
extmod/lib/mbedtls in the micro python repository has a copy of mbedtls;  I'm not sure what version, but it appears to be an older version that does not support TLS 1.3.   I'm not sure when/how the micropython team up-revs this.  You've reached the end of my knowledge on this issue...

m0ot, 2019-04-03:
can you please share here the link of extmod/lib/mbedtls because I coudn't find the file lib inside extmod file!
great thanks for sharing your knowledge!

dpgeorge, 2019-04-04:
@m0ot the mbedtls code used by the esp32 is part of the ESP IDF, in the directory components/mbedtls/mbedtls of the IDF (wherever you installed it).  The version of mbedtls currently used by the ESP IDF is 2.13.1, which doesn't support TLS 1.3.  As far as I know even the very latest version of mbedtls doesn't support TLS 1.3.

m0ot, 2019-04-04:
@dpgeorge Yes, you're right! https://tls.mbed.org/api/ as you just said the latest version of mbedtls does NOT support TLS 1.3.
I think the best way to get a lightweight TLS 1.3 is to try to add wolfssl in  IDF and MicroPython. Do you think it would be a good idea? and according to your experience, do you think it would require a lot of work hours to do that?

dpgeorge, 2019-04-04:
> I think the best way to get a lightweight TLS 1.3 is to try to add wolfssl in IDF and MicroPython. Do you think it would be a good idea?

It's certainly possible to do that, and it would be independent of the ESP IDF (ie just as easy to make it work for any port).  But keep in mind the license of wolfSSL (GPLv2, and commercial), meaning the library wouldn't be included here directly, although I don't see a problem having bindings to it for use by those who are OK with the licensing of wolfSSL.

The effort required depends on the experience one has with wolfSSL and with writing MicroPython bindings.