micropython/micropython #3152: unix: select.select is absent only on unix

ThunderEX, 2017-06-16:
when I'm a socket server, I find `select.select` is available on all targets(esp8266, stmhal, etc) except unix. since `select.select` is most cross-platform function within `select`, I think it makes much sense to support it.

dpgeorge, 2017-06-16:
select.select is useful for quick-and-dirty usage, but select.poll() is much more efficient, and lots of work went into the design of the ipoll() method so it doesn't need to use heap memory.  It's highly recommended to use ipoll(); it's available on all ports including unix.

pfalcon, 2017-06-16:
And this information is available in the documentation: http://docs.micropython.org/en/latest/pyboard/library/select.html#select.select . (Updated with https://github.com/micropython/micropython/commit/94696973a01059ffb370ada98f655b08de9b6302)

closed 2017-06-16 by pfalcon

ThunderEX, 2017-06-16:
but

- uselect is not available on windows port of micropython,
- ipoll is also not availble on cpython,

As documentation mentioned, select() is for compatible, select.select could be the only intersection available on both micropython and cpython & all platforms. Otherwise `select` on micropython and cpython are seemd to be two completely different modules.

And, there is no documentation mentioned ipoll()?

pfalcon, 2017-06-16:
> uselect is not available on windows port of micropython,

That's the problem of Windows port, patches welcome.

> Otherwise select on micropython and cpython are seemd to be two completely different modules.

They are - read http://docs.micropython.org/en/latest/pyboard/library/index.html . We can't pledge for full CPython compatibility - it's way too bloated. But we offer a path for compatibility in different direction, MicroPython -> CPython, which is, again, uselect.poll(). We also offer a way for interested parties to implement better CPython compatibility, that's why MicroPython's module is named "uselect", so you can implement "select" as you wish. There's even a sister, but separate project for that - micropython-lib. Nobody bothered to implement "select.select()" there yet though, which may give a hint how useful it is actually. Patches welcome.

dpgeorge, 2017-06-17:
> ipoll is also not availble on cpython,

To reiterate what @pfalcon said above, and summarise the situation:
- uselect.select() is available only on some ports, and is for quick-and-dirty usage
- uselect.poll() and the .poll() method of that object is available on all ports (maybe not uPy windows but that's to be implemented) as well as CPython (for most OS's) and is the recommended way to write portable code
- the .ipoll() method of the Poll object is uPy specific and is recommended for use when writing uPy-only code that should not use heap memory