micropython/micropython #3153: uasyncio: asynchronous context manager fails to run __aexit__() if return issued

peterhinch, 2017-06-16:
If a ``return`` statement is issued in an ``async with`` block, ``__aexit__()`` is not called. If the code simply runs to completion it works correctly. The code below never prints ``"In __aexit__()"``. Tested on Pyboard V1.0.

```python
import uasyncio as asyncio

class ExitGate():
    async def __aenter__(self):
        print('In __aenter__()')
        await asyncio.sleep_ms(0)

    async def __aexit__(self, *args):
        print('In __aexit__()')
        await asyncio.sleep_ms(0)

async def bar(exit_gate):
    async with exit_gate:
        for _ in range(5):
            print('Running bar()')
            await asyncio.sleep(1)
        return  # If this is commented out, __aexit__() runs

async def foo():
    exit_gate = ExitGate()
    loop = asyncio.get_event_loop()
    loop.create_task(bar(exit_gate))
    await asyncio.sleep(10)
    print('foo() complete.')

loop = asyncio.get_event_loop()
loop.run_until_complete(foo())
```
Outcome:
```
MicroPython v1.8.7-797-ge5e49be-dirty on 2017-06-10; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> import rats22_bug
In __aenter__()
Running bar()
Running bar()
Running bar()
Running bar()
Running bar()
foo() complete.
>>>  
```

dpgeorge, 2017-06-17:
I can confirm this is a problem.  But it looks like the root cause is an error in PEP492, see the description/spec of "async with" here: https://www.python.org/dev/peps/pep-0492/#asynchronous-context-managers-and-async-with

uPy implements what's written in the spec, using try/except/else to synthesise the async with statement.  But that's not how CPython behaves, CPython behaves as if it uses try/except/finally, to ensure that `__aexit__` is always executed (it's executed as part of the finally).

So first step would be to clarify with upstream CPython, see if PEP492 is in error, and go from there.

dpgeorge, 2017-06-17:
The CPython language docs also specify the try/except/else behaviour: https://docs.python.org/3/reference/compound_stmts.html#async-with

peterhinch, 2017-06-17:
So the CPython devs spotted the bug in PEP492, coded it correctly, and neglected to update the docs and the PEP. Given that CPython behaviour is as one would reasonably expect, is it not safe to follow that?

peterhinch, 2018-06-22:
This issue cropped up [in the forum](https://forum.micropython.org/viewtopic.php?f=2&t=4927&p=28299#p28299). In my opinion this issue should be addressed: MicroPython does not behave as per  reasonable expectations.

dpgeorge, 2018-06-22:
Thanks for digging up this issue.  Looking at the comments in the CPython code on how async with is really implemented, it states:
```
   The semantics outlined in that PEP are as follows:

   async with EXPR as VAR:
       BLOCK

   It is implemented roughly as:

   context = EXPR
   exit = context.__aexit__  # not calling it
   value = await context.__aenter__()
   try:
       VAR = value  # if VAR present in the syntax
       BLOCK
   finally:
       if an exception was raised:
           exc = copy of (exception, instance, traceback)
       else:
           exc = (None, None, None)
       if not (await exit(*exc)):
           raise
```
That's pretty much the same as a normal `with`, but just using `await` on the context handler functions.

Hopefully this can be implemented without modifying the bytecode.

dpgeorge, 2018-06-23:
Please see #3890 for a patch to fix this issue.

dpgeorge, 2018-06-27:
Now fixed by d8dc918deb8d4b13b8919706f9f208542c9ef2e6

closed 2018-06-27 by dpgeorge