micropython/micropython #3154: NodeMCU stating issue

MoonshineSG, 2017-06-20:
After a very uneventful  setup of the latest `esp8266-20170612-v1.9.1.bin` firmware on a new NodeMCU I am stuck. 
Using `esptool.py` v 2.0 on RaspberryPi, and `minicom` and after connection there is no prompt from MicroPython. A soft reset using the button, shows the screen below: 
 
```
dÜòý#4 ets_task(40100164, 3, 3fff829c, 4)
OSError: [Errno 2] ENOENT

MicroPython v1.9.1-8-g7213e78d on 2017-06-12; ESP module with ESP8266
Type "help()" for more information.
>>> 
```
(the funny characters in front are aways different...)

any user input has no response

dpgeorge, 2017-06-20:
Did you erase the flash completely before putting the MicroPython firmware on the board?

If you get the banner and the ">>>" characters then its working, and the lack of response when you type characters could be due to the board not working, the USB-UART connector may have a broken RX line.

MoonshineSG, 2017-06-20:
i did a `erase_flash` before uploading the firmware. 

If the RX line is broken, wouldnt  the `esptool.py` complain that it can't properly communicate and fail to load the firmware ?

dpgeorge, 2017-06-20:
> If the RX line is broken, wouldnt the esptool.py complain that it can't properly communicate and fail to load the firmware ?

Possibly.  But if you erase the flash, upload correctly and get the uPy ">>>" prompt, then the only real place things can go wrong is in the hardware.  I just checked the latest firmware (at 101fc187f009b8af74467f880e544f466881e853) and everything works OK for me on an Adafruit Huzzah Feather board.

MoonshineSG, 2017-06-20:
just to make sure communication and the board are not faulty, I uploaded an arduino sketch and everything seams ok... 

The board is one of these: http://www.hotmcu.com/nodemcu-lua-wifi-board-based-on-esp8266-cp2102-module-p-265.html

MoonshineSG, 2017-06-20:
I re uploaded the micropython firmware and, just for fun, i tried to open with the arduino serial monitor. And it worked. 
Back to `minicom` and check settings... `Hardware Flow Control ` was on... Once I turned it `off` it works! 

Sorry for the time wasted.

closed 2017-06-20 by MoonshineSG

dpgeorge, 2017-06-20:
Ok, thanks for reporting back.  I updated the title to reflect the true issue.

pfalcon, 2017-06-20:
@MoonshineSG : Please use forum http://forum.micropython.org/viewforum.php?f=16 in the future for such issues.

MoonshineSG, 2017-06-20:
if I knew what the issue is, i would not need the forum or this place... it looked like smth is wrong with mp... 

but it's good to know there's another place to get help...

juransaber, 2018-10-13:
I also have this problem in v1.9.4, usb chip is cp2102, but another board which use ch340 works well

N0rimaki, 2020-06-20:
> I re uploaded the micropython firmware and, just for fun, i tried to open with the arduino serial monitor. And it worked.
> Back to `minicom` and check settings... `Hardware Flow Control ` was on... Once I turned it `off` it works!
> 
> Sorry for the wasted time.

still important in 2020 (putty flowcontrol = none)