micropython/micropython #3155: modlwip: socket.read(n) on a socket with a finite timeout can lose data

dpgeorge, 2017-06-20:
During investigation of issue #2402 I came across the following problem with modlwip: if one creates a socket that has a finite timeout (eg `s.settimeout(5)`) and then does a read (eg `s.read(100)`) but the socket doesn't have that many available bytes, then `OSError(ETIMEDOUT)` is raised after the timeout has passed and any partial read (eg of 50 bytes) is completely lost.

Test script for server on PC (non uPy):
```python
import socket
s = socket.socket()
s.bind(('', 8000))
ss, _ = s.accept()
ss.send(b'1234')
```
Corresponding test script for modlwip (eg esp8266):
```python
import socket
s = socket.socket()
s.settimeout(5)
s.connect(('<ip of PC>', 8000))
s.read(100)
```
Outcome: the s.read() in the modlwip script raises ETIMEDOUT and the `b'1234'` data is lost.

The problem is that the stream reading function (in this case mp_stream_rw) is trying to read exactly 100 bytes.  It reads the 4 bytes (the "1234") and then tries to read 96 more.  But the modlwip socket fails after 5 seconds with ETIMEDOUT and the stream function raises this error, losing the 4 bytes it did read.

A solution would be to use stream_read1 instead of stream_read for the socket.

pfalcon, 2017-06-22:
In MicroPython, read()/write() are defined to be non-short (well, as you remember, it's actually up to a particular class, and for sockets, it's definitely non-short behavior, because short behavior is available via recv()/send()).

In this sense, the behavior is correct: someone calling timeout_sock.read(100) wanted one of: a) exactly 100 bytes; b) less bytes in case of EOF; c) error (exception), in particular when one couldn't read all of 100 bytes within timeout. The exception one got. The workaround is (should be, fingers crossed) to use send/recv.

You can also look up how upstream deals with that - throwing exception with partial data as an attribute. Hopefully, we don't need that.

P.S. We're entering a cluster of rabbit holes with issues like this and #3161.

dpgeorge, 2017-06-23:
> In this sense, the behavior is correct: someone calling timeout_sock.read(100) wanted one of: a) exactly 100 bytes; b) less bytes in case of EOF; c) error (exception), in particular when one [...]

We can't predict all uses cases and in this particular case it's not acceptable that read() loses data.  At the very least it should push the partially read data back so it can be read again (like what would happen in non-blocking mode).

> You can also look up how upstream deals with that - throwing exception with partial data as an attribute

For the example I give above, CPython will return straightaway with the short data.  That's really how uPy should behave for this case.

pfalcon, 2017-06-23:
> At the very least it should push the partially read data back

There's nowhere to push data back.

All I can say is:

https://docs.python.org/3/library/io.html#io.BufferedIOBase :

> In addition, those methods can raise BlockingIOError if the underlying raw stream is in non-blocking mode and cannot take or give enough data

The MicroPython behavior is compliant with this specification (modulo specific exceptions used, as we have simplified exception hierarchy).

dpgeorge, 2017-06-26:
> There's nowhere to push data back.

You can cache it in the socket object.  But of course that is not a very nice way to solve the problem.

> The MicroPython behavior is compliant with this specification

But, in my understanding, a socket with a timeout is *not* in non-blocking mode, and so those specs are not relevant in this case.  According to https://docs.python.org/3/library/socket.html#socket-timeouts a socket with a non-zero but finite timeout is different from both blocking and non-blocking modes.

Python introduced this notation of timeout on a socket (it's synthesised internally) and one option for MicroPython is to not implement finite timeout values, ie just support pure blocking and non-blocking.  That will simplify a lot of the socket code (for all ports, see that cc3200 and esp32 need to do special things to synthesise timeouts).

pfalcon, 2017-06-26:
> a socket with a non-zero but finite timeout is different from both blocking and non-blocking modes.

Indeed it is, that's an argument that it doesn't have to work like non-blocking, which I presented.

> and one option for MicroPython is to not implement finite timeout values, ie just support pure blocking and non-blocking.

That's why I was "cons" re: implementing it in Unix port. Well, I guess it's too late to remove it from existing implementations.

I guess we also should go and make changes you suggest (which makes timeout sockets handling more like non-blocking), though I hope you agree that my usecase above also makes sense (which treats timeout sockets more line blocking).