micropython/micropython #3156: Failure to compile 1.9.1 with threading module

kevster, 2017-06-20:
Hi,
I'm relatively new to MicroPython and this is my first time compiling firmware. I read in the 1.9.0 changelog that it should be possible to add the _thread-module for pyboards (I use the V1.1). I've compiled the latest 1.9.1 firmware and uploaded it to the pyboard successfully, but when I try to do so with the _thread-module, it fails to compile. What am I doing wrong?

I've added the following to /stmhal/mpconfigport.h as per the changelog instructions:
```
#define MICROPY_PY_THREAD           (1)
#define MICROPY_PY_THREAD_GIL       (1)
```

Result from 'make BOARD=PYBV11':

```
...
CC ../py/objmap.c
CC ../py/objmodule.c
In file included from ../py/mpstate.h:35:0,
                 from ../py/objmodule.c:30:
../py/objmodule.c:169:19: error: 'MP_QSTR__thread' undeclared here (not in a function)
     { MP_ROM_QSTR(MP_QSTR__thread), MP_ROM_PTR(&mp_module_thread) },
                   ^
../py/obj.h:91:56: note: in definition of macro 'MP_OBJ_NEW_QSTR'
 #define MP_OBJ_NEW_QSTR(qst) ((mp_obj_t)((((mp_uint_t)(qst)) << 2) | 2))
                                                        ^~~
../py/objmodule.c:169:7: note: in expansion of macro 'MP_ROM_QSTR'
     { MP_ROM_QSTR(MP_QSTR__thread), MP_ROM_PTR(&mp_module_thread) },
       ^~~~~~~~~~~
make: *** [build-PYBV11/py/objmodule.o] Error 1
```

kevster, 2017-06-20:
Argh, beginner's mistake. It seems my initial compile, without the _thread-module enabled hadn't been cleaned. After deleting the entire build-PYBV11-directory and recompiling it works like a charm! Sorry for the inconvenience.

closed 2017-06-20 by kevster

dpgeorge, 2017-06-21:
> After deleting the entire build-PYBV11-directory and recompiling it works like a charm!

So the problem here looks the same as #2982.