micropython/micropython #3157: Unix: Unexpected behaviour in nested for loops

peterhinch, 2017-06-21:
This arose in [this forum post](https://forum.micropython.org/viewtopic.php?f=15&t=3490&p=20344#p20344)
```python
for foo in range(10):
  print(foo)
  for bar in range(10):
    pass
  else:
    continue
  break
```
Outcome under CPython (prints 0-9) and MicroPython (prints 0 only)
```
[adminpete@axolotl]: ~
$ upython 
MicroPython v1.8.7-797-ge5e49be-dirty on 2017-06-05; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import rats26
0
>>> 
[adminpete@axolotl]: ~
$ python3.5
Python 3.5.2 (default, Dec  2 2016, 15:29:25) 
[GCC 4.8.4] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rats26
0
1
2
3
4
5
6
7
8
9
>>> 
```

dpgeorge, 2017-06-21:
Thanks for the report, I can confirm that it's a bug with the optimised for loop.  For now, to work around it force the compiler to not generate optimised loops by using an explicit iterator:
```python
xrange = range
for foo in range(10):
  print(foo)
  for bar in xrange(10):
    pass
  else:
    continue
  break
```

pfalcon, 2017-06-21:
@dpgeorge : My understanding is that it's incorrect handling of "continue" scoping in for's "else" clause: it no longer applies to that "for", but to the enclosing "for".

dpgeorge, 2017-06-22:
Should be fixed by 4c5f108321a8fd3f67f597ca918427eda813c12e.  Tests added in 44922934f55cb1cb8a64eba4afabb66563d66349

dpgeorge, 2017-06-22:
> My understanding is that it's incorrect handling of "continue" scoping in for's "else" clause: it no longer applies to that "for", but to the enclosing "for".

No, the else is handled correctly.  It was an issue with values remaining on the Python stack when they should have been popped.

closed 2017-06-22 by dpgeorge