micropython/micropython #3158: py: bytes() won't call obj.__bytes__()

ThunderEX, 2017-06-21:
I defined a class with __bytes__ method. Calling bytes() on its instance should redirect to its __bytes__ method, but it doesn't.

Can this be fixed?

pfalcon, 2017-06-21:
> Calling bytes() on its instance should redirect to its bytes method, but it doesn't.

To set the expectations right: it should in CPython, but not necessarily in MicroPython. MicroPython implements small subset of CPython functionality, and that's by design. In more detail, "zen" of MicroPython is described here: https://github.com/micropython/micropython/wiki/ContributorGuidelines

For your case you can workaround the issue by calling your `__bytes__()` method instead of `bytes()` constructor:

~~~
a = bytes(my_obj)
~~~
=>
~~~
a = my_obj.__bytes__()
~~~

Such code will be compatible with CPython.

kevinjwalters, 2019-04-06:
I've just found this out the hard way too on CircuitPython 4.0.0-beta.5 which is based on MicroPython 1.9.4 @25ae98f. It's not at all obvious from the error message what is going on here. Can the error message be improved?

```
>>> class onetwo():
...     def __init__(self):
...         self.data = [1, 2]
...     def __bytes__(self):
...         return bytes(self.data)
...
>>>
>>> obj1 = onetwo()
>>> bytes(obj1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'onetwo' object is not iterable
```

If you run unit tests like this on Python 3 they'll pass and you get a surprise when it gets deployed on micropython :(

dpgeorge, 2019-04-08:
> Can the error message be improved?

Making better error messages is sometimes difficult and increases code size as well.  For the case here, it's probably cheaper (in code size) to actually support delegating to `__bytes__`.  (Well, implementing it would still leave the confusing error message for other unsupported objects.)

> If you run unit tests like this on Python 3 they'll pass and you get a surprise when it gets deployed on micropython

It's highly recommended to test code using a PC version of MicroPython, so incompatibilities like this can be found more quickly.  There are a lot of small details that are different between Python and MicroPython, simply due to the nature of the language and how MicroPython is implemented.