micropython/micropython #3160: finaliser is not enabled in esp8266 port

cwt, 2017-06-22:
I know that user object finaliser (`__del__`) will never be called in micropython. However, during my experiment with `urequests` lib, I found that the esp8266 port didn't enable the finaliser. So I compiled my firmware from git with just this one line changed.

```diff
$ git diff
diff --git a/esp8266/mpconfigport.h b/esp8266/mpconfigport.h
index 483f93d..0f7ff94 100644
--- a/esp8266/mpconfigport.h
+++ b/esp8266/mpconfigport.h
@@ -17,6 +17,7 @@
 #define MICROPY_DEBUG_PRINTER_DEST  mp_debug_print
 #define MICROPY_READER_VFS          (MICROPY_VFS)
 #define MICROPY_ENABLE_GC           (1)
+#define MICROPY_ENABLE_FINALISER    (1)
 #define MICROPY_STACK_CHECK         (1)
 #define MICROPY_ENABLE_EMERGENCY_EXCEPTION_BUF (1)
 #define MICROPY_KBD_EXCEPTION       (1)
```

With this finaliser-enabled firmware, I can make many requests using urequests lib without any error with no need to call `response.close()` explicitly. And the free resource after I stop my program seem to be more available than the firmware without finaliser.

Freemem on firmware **without finaliser** after I make 5 http requests (after that it just stop working):
```python
>>> import esp
>>> esp.freemem()
7728
```

Freemem on firmware **with finaliser** after I make 10 http requests without any error:
```python
>>> import esp
>>> esp.freemem()
8608
```

So should we enable `MICROPY_ENABLE_FINALISER` on esp8266 by default or is there any disadvantage that I didn't aware of?

dpgeorge, 2017-06-22:
> So should we enable MICROPY_ENABLE_FINALISER on esp8266 by default or is there any disadvantage that I didn't aware of?

I'm kind of surprised that it's not already enabled, especially since resources are so tight anything that can be done to help in that area is welcome.  The only disadvantage is increase in firmware size.

> Freemem on firmware without finaliser after I make 5 http requests (after that it just stop working):

To clarify: when you had the finaliser disabled, did you explicitly close your responses from urequests?

cwt, 2017-06-22:
@dpgeorge  I just delete response objects on both firmwares. If I call response.close() after all requests, my program survived on firmware without finaliser. So, this make me surprise too that by enable finaliser and without closing the response, my program still working after more than 1000 loops (and still continue working now).

dpgeorge, 2017-06-22:
> If I call response.close() after all requests, my program survived on firmware without finaliser.

Ok, so that means your code can run for a very long time without finaliser, by making sure you do a close?  Then what exactly prompted you to enabled the finaliser?

> So, this make me surprise too that by enable finaliser and without closing the response, my program still working after more than 1000 loops

Well, that part is not so surprising, the finaliser code should work and should reclaim sockets (a socket object has a `__del__` which is called by the finaliser; but the response object itself won't be called, it doesn't need to be).

cwt, 2017-06-22:
Long story: at first I tried to investigate why my code stop working after 5 http requests, so I looked at urequests code and saw the [example](https://github.com/micropython/micropython-lib/blob/master/urequests/example_xively.py) stated that I must close the response by calling `response.close()`. Without any knowledge about micropython finaliser before, I just thought that why the lib didn't just put `self.close()` inside `__del__`, so I did change urequests code, and to my surprise that micropython never call my `.__del__()` when I delete response. I though it was a bug, so I searched the GitHub issues for `__del__` and found #1878. For my curiosity, I really wanted to know what exactly was finaliser, what I did next was `$ grep -rin finaliser .` and then saw the line `#define MICROPY_ENABLE_FINALISER`  that was enabled/disabled on some ports, but that line is missing on esp8266. I just wonder if the finaliser was enabled, what will happen to my code.

pfalcon, 2017-06-22:
@dpgeorge : During discussion of BT implementation of problems on ESP32, there was a claim (from me) that that there should be either explicit tracking of objects passed to 3rd-party libs (with references stored outside of upy heap), or finalization (so, if upy thinks that an object is dead, this fact was also communicated to 3rd-party lib). So, for modlwip, finalizer should be always enabled, and if it's surprisingly not, that the firsts suspect of random crashes and potential wrong mp_obj_t values we were seeing.

pfalcon, 2017-06-22:
I guess we didn't enable finalizer initially to save memory, and then it followed that way without anyone checking.

pfalcon, 2017-06-22:
@cwt: Mind that regardless of your experiments, you still should call .close() explicitly on any object, and as soon as you finished working with it - it's not guaranteed when finalizer will be called (or that it will be called at all, as you've seen).

> 7728
> 8608

These numbers without full context don't mean much. Are you familiar with modulo arithmetic?

~~~
(10 + 10) mod 30 = 20
(10 + 10 + 10) mod 30 = 0
~~~

From the above, you can't conclude that 30 is less than 20, nor from your random test results you can conclude that finalizer is more efficient for memory - it's definitely not, as its enabling takes useful memory away from user apps. But it needs to be enabled because it's not safe to not enable it.

cwt, 2017-06-22:
@pfalcon noted and thank you for your explanation.

pfalcon, 2017-06-23:
Enabled in 0fe825b89e9c6a2a0e379ca61156f8b47f56b333, testsuite is ok. This decreased usable heap size by 320 bytes.

@cwt : Thanks for the report.

closed 2017-06-23 by pfalcon