micropython/micropython #3161: Missing API to connect() using nonblocking Sockets

simon-budig, 2017-06-22:
On the usual BSD-Socket API you can set up nonblocking sockets for connect()ing to an external server. The general API is described e.g. in this stackoverflow message: https://stackoverflow.com/questions/17769964/linux-sockets-non-blocking-connect

Using nonblocking sockets for a connect() using micropython seems to work, providing everything works out OK. But the API to check the status of the socket (i.e. getsockopt()) is missing, it is unclear how one is supposed to check for errors in the poll() when the connection becomes writable...

Please add this API or extend the documentation on how to do this. Thanks!

pfalcon, 2017-06-22:
You deal with that case for connect() the same way as you deal with such case for any other non-blocking socket function. For example, how do you know if send() succeeded? Simple - if it didn't, you'll get an error on a call to a next socket function.

The reference implementation of non-blocking socket handling is uasyncio module from micropython-lib project, feel free to refer to it.

simon-budig, 2017-06-22:
so what is a no-op socket function I could call to test if the connect() succeeded? I can't use send() since I don't have any data to send yet...

pfalcon, 2017-06-22:
You don't have to call any special function. Just use sockets "as usual". If there's error, you'll know about it via exception.

Otherwise, if you're concerned with this case, feel free to provide a realistic, reproducible usecase when it may be a real issue.

simon-budig, 2017-06-22:
Ok, here is a scenario.

I want a client to connect to a server, where it expects data from the server and does not send any data to the server. The connect() needs to be done nonblocking, since (e.g.) a LED is supposed to blink during the initialisation of the connection.

It is desirable to know as fast as possible if the server is unreachable or the connection is refused.

The following code builds a connection to a server but uses the wrong port (i.e. the server resets the connection). However, the code can not detect that the connection is failed, since it can't query the error-state of the socket (what is supposed with the BSD-variant of the API).

The code idles around, waiting for the socket to become readable, thinking that the connection to the server has been built correctly and the server just not sending any commands.

In reality the connection has failed, the server has no chance of sending any commands.

```
MicroPython v1.9.1-8-g7213e78d on 2017-06-12; ESP module with ESP8266
Type "help()" for more information.
>>> import polltest
>>> polltest.do_client ()
connecting to "esptest"
...done
building connection to ('192.168.4.1', 81)
connect [Errno 115] EINPROGRESS
Event: (<socket state=1 timeout=0 incoming=0 off=0>, 4)
new connection
connection socket: <socket state=1 timeout=0 incoming=0 off=0>
```

This is the code in question:

```python
import network, time
import socket, select

def do_client (port=81):
   sta_if = network.WLAN (network.STA_IF)
   ap_if = network.WLAN (network.AP_IF)
   sta_if.active (False)
   ap_if.active (False)
   time.sleep (0.05)
   sta_if.active (True)
   sta_if.connect ("esptest", "12345678")
   print ("connecting to \"esptest\"")
   while not sta_if.isconnected ():
      time.sleep (0.05)
   print ("...done")

   p = select.poll ()
   addr = socket.getaddrinfo (sta_if.ifconfig ()[-1], port)[0][-1]
   print ("building connection to", addr)

   sock = socket.socket ()
   sock.setblocking (False)
   p.register (sock, select.POLLERR | select.POLLOUT | select.POLLIN | select.POLLHUP)
   try:
      sock.connect (addr)
      print ("connect done")
   except OSError as e:
      print ("connect", e)

   while True:
      for event in p.ipoll (-1):
         print ("Event:", event)
         if event[1] & select.POLLOUT:
            print ("new connection")
            print ("connection socket:", event[0])
            p.modify (event[0], select.POLLERR | select.POLLIN | select.POLLHUP)
         elif event[1] & select.POLLIN:
            data = event[0].read (32)
            if data:
               print ("got data", repr (data))
            else:
               print ("short read, closing connection")
               p.unregister (event[0])
               event[0].close ()
         elif event[1] & select.POLLHUP:
            print ("HUP, closing connection")
            p.unregister (event[0])
            event[0].close ()
         else:
            print ("unknown event:", event)
            p.unregister (event[0])
            event[0].close ()
```

dpgeorge, 2017-06-23:
It might be possible to do what you want by using the POLLERR/POLLHUP flags.  Running a simpler version of your script (see related #3164) shows that, when the connection is refused, CPython and uPy unix both return an event from poll that has all the flags set (IN, OUT, ERR, HUP).  So if you just check for HUP/ERR as the first part of the if block, and IN/OUT at the end, then it might work.  Except that esp8266 doesn't set the HUP or ERR flags, so it won't work, but at least in principle that's how it could be done without using getsockopt().

simon-budig, 2017-06-23:
Maybe there is an argument to be made here, that there should be an API available to retrieve the actual error, when poll returns with the POLLERR flag set. Using send() or read() in that case seems wrong (we don't know if the socket is readable or writable, so reading and writing might double up on the error...).

I don't know. Implementing getsockopt() does not strike me as an expensive thing here, but I am new to micropython, so I don't know if there are conceptual arguments against doing that...

pfalcon, 2017-06-23:
> Except that esp8266 doesn't set the HUP or ERR flags

Then it's https://github.com/micropython/micropython/issues/3073

pfalcon, 2017-06-23:
> I don't know. Implementing getsockopt() does not strike me as an expensive thing here

Then you're welcome to try. MicroPython has 4-5 socket module implementations for different ports, of course all they need to be supported (actually, all future too). Then there should be regression tests to ensure they work as expected (on all socket implementations above), and keep working in the future (https://github.com/micropython/micropython/issues/3014).

pfalcon, 2017-06-23:
> The code idles around, waiting for the socket to become readable, thinking that the connection to the server has been built correctly and the server just is not sending any commands.

This should not happen, polling a socket in error state for "readable" status should return true, so read can be issued, ending in error.

jonnor, 2024-09-29:
There was a suggestion to use POLLERR/POLLHUP flags for this. Can this be closed, now that https://github.com/micropython/micropython/issues/3073 has been fixed?