micropython/micropython #3162: ESP8266 RTC.memory() boundary case

peterhinch, 2017-06-22:
This only fails if p == 492. Smaller values work. Larger values throw a ``ValueError: buffer too long``. 
```python
import machine
rtc = machine.RTC()
p = 492
a = bytes((n & 0xff for n in range(p)))
rtc.memory(a)
for k in range(p):
    if a[k] != rtc.memory()[k]:
        print(k, a[k], rtc.memory()[k])
```

pfalcon, 2017-06-22:
Throws where?

dpgeorge, 2017-06-23:
The throwing of a ValueError in the first rtc.memory(a) is correct, when a is too large to fit in the RTC memory.

The issue reported here (which I can confirm) is that, after storing 492 bytes to the RTC memory, reading the data back gives incorrect values.  I think the bug in the code is the formula which is aligning the length of the data to 4 bytes.  In the case that the length is already a multiple of 4 bytes the formula adds another 4 bytes, so this will overflow when you are already at the maximum value (ie 492).

dpgeorge, 2017-06-26:
Fixed by caa132a236f12380dfa673adf8f9bdad0051f799.

closed 2017-06-26 by dpgeorge