micropython/micropython #3168: OneWire: Support internal pull-up resistor

catdog2, 2017-06-23:
I'm playing around with an esp8266 and a ds18b20 sensor. I noticed that the _OneWire_ class calls the init function of the pin inside its _\_\_init\_\__ [1] function which disables a previously set _pull=Pin.PULL_UP_. I can get it to work by calling _ow.pin.init(Pin.OPEN_DRAIN, pull=Pin.PULL_UP)_ with _ow_being an instance of _OneWire_ but this is hardly a nice API.

[1] https://github.com/micropython/micropython/blob/a065d786753b83ab92873b90d8d61d9fe5639fdd/drivers/onewire/onewire.py#L17

dhylands, 2017-06-23:
Interesting. The same bug existed in the earlier version of the lib, and I fixed it here:
https://github.com/micropython/micropython/commit/3c84197f17d27a991f50cb6e5ed268edbc796991#diff-a0ac4c7ae6cbf3b0256fcae26704fb4c

It looks like the "new" version went and reverted that behaviour.

dpgeorge, 2017-06-26:
There were actually two 1-wire drivers, one for pyboard and one for esp8266.  The pyboard one was modified to use PULL_UP, but the esp8266 one never had it.  And very recently the esp8266 driver (which was more efficient, minimal and generic) replaced the pyboard version; see a065d786753b83ab92873b90d8d61d9fe5639fdd.

I agree that using PULL_UP in the pin initialisation is useful, and worked well on the pyboard, so it's been fixed/added in 683df1c8d5ab37dbb0d2909bcd94c7e4eea3232f

closed 2017-06-26 by dpgeorge