micropython/micropython #3171: Import Error: no module named 'stm'

athelopuce, 2017-06-25:
When I try to import stm module I have an importation error.
    Error: no module named 'stm'   
Is stm module enable for esp8266 as documentation indicates ?
Thanks

I use :  MicroPython v1.9.1-8-g7213e78d on 2017-06-12; ESP module with ESP8266  but it was the same with 1.9 version

pfalcon, 2017-06-25:
> Is stm module enable for esp8266 as documentation indicates ?

Where documentation indicates that?

athelopuce, 2017-06-25:
Here : 
[https://docs.micropython.org/en/latest/esp8266/reference/speed_python.html#accessing-hardware-directly](url)
thanks

pfalcon, 2017-06-25:
Well, if you read the section you link to, it starts with saying "Consider the example of toggling an output pin on the Pyboard.". So, no, "stm" module is not available on esp8266, it's for different CPU. In a926119099bae885a8f1f031c2cc4eacbf785df2, I updated that doc to make this fact clearer and make the doc more hardware neutral. (It will be on the docs site tomorrow.)

closed 2017-06-25 by pfalcon