micropython/micropython #3177: Allow forwarding between AP and STA interfaces

jadolg, 2017-06-29:
I oppened a question some days ago about it and have not found anything anywhere https://stackoverflow.com/questions/44781305/micropython-nat-between-interfaces-on-esp8266
turns out that I can connect to a hotspot and act as a hotspot at the same time, but forwarding among interfaces is not permitted. I've being redirected to other software like NodeMCU but I would really like to use Micropython (and not having to compile ti myself)

dpgeorge, 2017-06-30:
Such a feature is nice to have but is quite advanced and not really anything to do with uPy itself.  Rather it's about the lwip stack and the low-level wifi interface.  An initial investigation would be needed to see if it's possible within the constrains of the esp8266, if/how it's been implemented elsewhere, and what's needed to bring that to this code base.

pfalcon, 2017-06-30:
... then someone will need to test it, and make sure it can be tested in (semi)automated manner, then someone will need to support it over time (patiently answer questions like "nothing works? what's wrong!"). Interested parties are definitely welcome to start on that...

jadolg, 2017-06-30:
there are some implementations on lwip refered to me on the comments like [https://github.com/martin-ger/esp_wifi_repeater](https://github.com/martin-ger/esp_wifi_repeater) and [https://github.com/martin-ger/esp-open-lwip/tree/nat](https://github.com/martin-ger/esp-open-lwip/tree/nat) . Also some othe links refgering to NodeCMU firmware but nothing on micropython wich is what I want to use to develop my system.

pfalcon, 2017-06-30:
MicroPython uses this lwip repo: https://github.com/pfalcon/esp-open-lwip (normally, via https://github.com/pfalcon/esp-open-sdk). You're welcome to hack it as you like, then clean it up and share results (with tests proving that what you did really works).

jadolg, 2017-06-30:
@pfalcon sadly my C background is very poor and I've never worked with this kind of things before. I'm a total newbie here. I would really like to see this feature incorporated to micropython tho so I'll just try to get the people who can actually code it together.

jadolg, 2017-07-07:
@pfalcon would it be too painful to merge this function from https://github.com/martin-ger/esp-open-lwip 
I believe that's where the juice is

jadolg, 2018-07-21:
Any news about this? I would really like to have this feature.

xplshn, 2025-04-07:
> [@pfalcon](https://github.com/pfalcon) would it be too painful to merge this function from https://github.com/martin-ger/esp-open-lwip I believe that's where the juice is

Indeed, I would appreciate this a lot too