micropython/micropython #3178: extmod: ure incorrectly handles literal '-' in character classes

alex-robbins, 2017-06-29:
According to [the CPython docs](https://docs.python.org/3/library/re.html#regular-expression-syntax), in a character class,
>  If `-` is escaped (e.g. `[a\-z]`) or if it’s placed as the first or last character (e.g. `[a-]`), it will match a literal `'-'`.

---
The first assertion statement below succeeds, but the next four fail. As indicated, some of the failures are AssertionErrors (incorrect result was returned) and some are ValueErrors (input was rejected, no value returned).

```python
import ure

# Passing:
assert ure.compile(r'[-a]').split('foo-bar') == ['foo', 'b', 'r']

# AssertionError:
assert ure.compile(r'[a\-x]').split('foo-bar') == ['foo', 'b', 'r']
assert ure.compile(r'[\-ax]').split('foo-bar') == ['foo', 'b', 'r']

# ValueError: Error in regex
assert ure.compile(r'[ax\-]').split('foo-bar') == ['foo', 'b', 'r']
assert ure.compile(r'[a-]').split('foo-bar') == ['foo', 'b', 'r']
```

All 5 of these work fine in CPython with `re` and in MicroPython with re-pcre.

alex-robbins, 2017-06-30:
@pfalcon,
I just read [your comment](https://github.com/micropython/micropython/issues/3179#issuecomment-312207169) on the other issue. 

I hadn't realized that re1.5 is a separate project (I assumed that ure was strictly a MicroPython-specific regex implementation). Should those issues (this one and #3176) be reported against the re1.5 repository? Should I avoid reporting re1.5 issues here in the future?

pfalcon, 2017-06-30:
@alex-robbins : The big question is whether you're ready to go beyond bug reports. If you do, then if you hit bugs with MicroPython, it makes sense to report them for MicroPython, but as you noticed, fixing them will be in re1.5, which is a separate project with (potentially) separate requirements, so way to fix them (if at all) may require separate discussion in a re1.5 ticket.

If you're not ready to go further, then reporting here is probably better, as uPy more popular project (but we don't have lack of ticket as you noticed, so it may take (quite) some time before the maintainers get to lower-priority tickets after higher-priority).

alex-robbins, 2017-06-30:
In general I might be willing to contribute code, but in the specific case of messing with a regex implementation, I'm somewhat reluctant, because that sounds hard. :smile:

I guess I'll continue to report anything new here, unless I know beforehand that I want to provide a patch. Thanks.

pfalcon, 2017-09-24:
Fixed in 4a314a6f63c7157a824aa98c01009a8fbafb7475.

dpgeorge, 2017-10-05:
Note that the original bug report here is not fully fixed, escaped hyphens within a character class do not work.  @pfalcon why did you change the title of the issue to reflect only half of the bug report?

pfalcon, 2017-10-05:
I did not change it "to reflect only half of the bug report". The original title claimed that "ure incorrectly handles literal '-' in character classes", and that was incorrect, because `[-abc]` worked well.

dpgeorge, 2017-10-05:
The original post explicitly states that escaped hyphens (literal hyphens) are not supported, and gives 3 example code snippets to test these cases.  What was the reason to ignore that part of the issue?

pfalcon, 2017-10-05:
@dpgeorge : Thanks for asking so thoroughly. One possible reason that there was a separate report/patch on that. Another reason is a golden rule "one issue per bug report". But in all fairness, I don't remember for sure. What's the problem anyway? If you think this should be reopened, please do. If you think that the title needs correction, please do.

pfalcon, 2017-10-05:
@dpgeorge : Thanks for asking so thoroughly. One possible reason that there was a separate report/patch on that. Another reason is a golden rule "one issue per bug report". But in all fairness, I don't remember for sure - we have 300+ open bugs. What's the problem anyway? If you think this should be reopened, please do. If you think that the title needs correction, please do.

dpgeorge, 2017-10-06:
> What's the problem anyway?

Well, I just wanted to be clear on what the fate of this issue was.  I found that there weren't any regression tests for the bug reported here (the tests are upstream in re1.5, but nothing here) and went to add them by using the code given in the first post here.  And then found that they still failed.  So I added tests just for the part that was fixed (see 98dd126e9818309e4ec9b561e7b6b9f7fa039cb9), and then wanted to understand why this issue was closed.

So the remaining issue is that `\-` is not handled in character classes.  But adding support for that would open up the question of why not also supporting other escape sequences; eg in CPython you can do `re.match(r'[\t]', '\t')` but that's not supported in uPy.  It doesn't really make sense to add support just for escaped hyphen, and adding support for all (or most) escape chars adds a lot of code so I'd say is a "won't fix" for the time being.

In summary: it's possible to specify literal hyphens in character classes as the first or last char.  But escape sequences within character classes are currently not supported and won't be unless it can be done with very minimal code increase.

pfalcon, 2017-10-06:
> Well, I just wanted to be clear on what the fate of this issue was.

Ok, let's trace. For this issue, 2 patches were posted, rightly separated. You rightly pointed that they should be submitted upstream too re1.5 project. Yes, ideally we'd have tickets open until issue is "completely" resolved, but that generally requires a dedicated QA engineer. You yourself closed recently few tickets which I kept as reminders for future work. This ticket got closed too given that there active patches.

There were not much activity on those patches re: review comments, so the maintainer (myself) looked into one https://github.com/pfalcon/re1.5/pull/30, reworked it, added tests, merged and upgraded downstream.

> (the tests are upstream in re1.5, but nothing here). So I added tests just for the part that was fixed (see 98dd126)

Right, I didn't add test to MicroPython because they're upstream, it's kind of duplication of work.

> So the remaining issue is that \- is not handled in character classes. [...]

Yes, so there's a pending patch on that, https://github.com/pfalcon/re1.5/pull/31, with the concern raised that it adds too much code to the re1.5 project, whose aim was to provide the minimal-size regex implementation, as inherited from the original re1 project. I hope the logic is clear here, and it's the same as for the MicroPython itself - it didn't make sense to create and put all the effort into development and maintenance of a project to make it lose its identity at the arbitrary point of time later, identity being the minimality. So, if there's desire to have more features, they should be added in a way to preserve the baseline. Given that both patches were from the same developer, I'd think that changes done to https://github.com/pfalcon/re1.5/pull/30 serve as a good example for https://github.com/pfalcon/re1.5/pull/31, but for avoidance of doubt, I posted explicit suggestions: https://github.com/pfalcon/re1.5/pull/31#issuecomment-334846591

Summing up: I'm looking forward to have that issue resolved. Eventually. But for that, someone will need to do further work. And for quite awhile (a year maybe), I don't think that the approach that maintainers [rush and] clean up after contributors is scalable and try to raise the level of contributions to MicroPython. That might be not work that well (so far), but that's how most (high-quality) open-source projects work - it's part of the social contract that it's contributors' responsibility to get their patches into shape suitable for upstream (the key of course "contributors'", not just "contributor's", so people can get together on that task). So, we can wait for someone to work on that. (Even though we know who's that someone usually keeps being still so far.)

In the meantime, let's reopen this for avoidance of doubt.

alex-robbins, 2017-10-09:
Updating my original comment:

```python
import ure

# Passing:
assert ure.compile(r'[-a]').split('foo-bar') == ['foo', 'b', 'r']

# Passing (used to be ValueError: Error in regex):
assert ure.compile(r'[a-]').split('foo-bar') == ['foo', 'b', 'r']
assert ure.compile(r'[ax\-]').split('foo-bar') == ['foo', 'b', 'r']

# AssertionError:
assert ure.compile(r'[a\-x]').split('foo-bar') == ['foo', 'b', 'r']
assert ure.compile(r'[\-ax]').split('foo-bar') == ['foo', 'b', 'r']
```

So, interestingly, the backslash-escaped `-` works at the end of the character class, but not in the middle or at the beginning. Also note that `ure` does not fail fast at compile time, but just gives the wrong answer in these cases.

---

Thanks, @deshipu for the patches.

deshipu, 2017-10-09:
Here's a crazy idea that might inspire someone: it would be possible to have the parser for the regular expressions actually use a regular expression (which would make it much easier to handle such cases), and use the code already in re to parse, if only the regular expressions for the parser itself were stored in the program in an already compiled form (there would be an additional step required for generating that with some ad-hoc parser like the current one). That could also potentially make the whole thing smaller.

dpgeorge, 2019-10-18:
The remaining cases were fixed in ebf8332104c1c6fbd01b521a713e49d040c62920

closed 2019-10-18 by dpgeorge