micropython/micropython #3179: extmod: ure's regex.split returns strings when given bytes

alex-robbins, 2017-06-29:
When given a bytes, regex.split should return a list of bytes. CPython and MicroPython with re-pcre both behave this way, but MicroPython's ure incorrectly returns a list of strings.

```python
MicroPython v1.9.1-13-g1e70fda6 on 2017-06-15; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import ure
>>> import re # re-pcre
>>> ure.compile(b'x').split(b'fooxbar')
['foo', 'bar']
>>> re.compile(b'x').split(b'fooxbar')
[b'foo', b'bar']
```

dpgeorge, 2017-06-30:
uPy also always returns strings for search and match, even if passed a bytes object, whereas CPython returns bytes.

IMO these should be fixed: there is correct handling of such things elsewhere, eg in str.split() and os.listdir(), and it hurts compatibility with CPython trying to work around the issue.

pfalcon, 2017-06-30:
I agree, and it would be in my list to fix. This is also the one truly uPy level issue, others (https://github.com/micropython/micropython/issues/3176, #3178) are re1.5 issues.

pfalcon, 2017-06-30:
Should be fixed by 58b7b01cb539cb23109faadaf373e0134b81a6da.

closed 2017-06-30 by pfalcon

alex-robbins, 2017-06-30:
Looks good to me. Thanks!