micropython/micropython #3183: Provide string ljust / rjust

dbrgn, 2017-06-30:
I just noticed that .ljust and .rjust are missing on the string object, while .center is available.

I saw that the .center method is behind the `MICROPY_PY_BUILTINS_STR_CENTER` build flag. Would you consider a pull request implementing ljust and rjust, possibly behind a `MICROPY_PY_BUILTINS_STR_LRJUST` flag?

pfalcon, 2017-06-30:
What usecases do you have for them? ljust/rjust are more or less trivially to implement in Python. center() is a bit less trivial, it was added from the concern of minimizing memory allocations. Adding behind config options is possible, but are you ok that it will be disabled by majority of ports?

dbrgn, 2017-06-30:
> What usecases do you have for them?

Alignment in serial console logs. I want that all log keys have the same width so that it's easy to process visually.

In Python I'd implement it like this:

```python
ljust = 8
context + ' ' * max(0, ljust - len(context))
```

I think a C implementation could be very small but more efficient than that. If the width value is smaller or erqual to the length of the current string, then no allocation needs to be done at all.

pfalcon, 2017-06-30:
Exactly, it's a simple one-liner in Python. In C, it will be couple hundreds of bytes. So, I'm -0.5 on that, it doesn't help with things like https://github.com/micropython/micropython/issues/2908 and other code size increases we have in queue to increase (much bigger) memalloc efficiency.

dbrgn, 2017-06-30:
Hm, a couple hundreds of bytes of binary size sounds a big large to me. Wouldn't it just be this (in pseudocode)?

```
if width <= length:
  return original_string
new_string = allocate_string(width)
copy_string(original_string, new_string)
write(fill_char, new_string + len(original_string), len(new_string))
```

I'm not really familiar with the micropython internals though.

robert-hh, 2017-07-01:
Why don't you use string formatting with the % operator or str.format(), especially in output to be read by humans? It's meant for that.

dbrgn, 2017-07-01:
@robert-hh oh, you are right! I wasn't aware that the formatting language supports that feature.

```python
>>> '{:<8} : hello'.format('foo')
'foo      : hello'
```

In that case there are probably very few use cases left for .ljust / .rjust :)

closed 2017-07-01 by dbrgn

dpgeorge, 2017-07-02:
For the record, % formatting also supports left and right justification: `'%-20s' % 'foo'` and `'% 20s' % 'foo'` respectively.

pfalcon, 2017-07-09:
This is now documented in https://github.com/micropython/micropython/commit/ad3abcd324cd841ffddd5d8c2713345eed15f5fd. Generally, affected parties are encouraged to submit such diffcases with workarounds.