micropython/micropython #3188: docs: Consistent use of articles in the top-level ToC headings

pfalcon, 2017-07-01:
Looking at http://docs.micropython.org/en/latest/esp8266/, why can see things like below in left-hand ToC pane:

~~~
MicroPython libraries
The MicroPython language
~~~

Let's be consistent with articles usage. It's a common practice in English to drop articles in headings, and majority of our headings follow that, so that's what I propose to apply consistently.

pfalcon, 2017-07-01:
Likewise: consistent word casing for titles. There's a convention to capitalize each word in English headings, but we don't follow it, then "MicroPython Differences from CPython" is the only inconsistency in top-level ToC, fixing it right away.

pfalcon, 2017-07-01:
Note that CPython appears to use articles in top-level headings, but mostly doesn't use in lower-level headings, e.g. https://docs.python.org/3/reference/index.html

dpgeorge, 2017-07-02:
> MicroPython libraries 
> The MicroPython language

I agree that shorter headings are neater, but in this case it really changes the meaning, viz:
- "The MicroPython language" specifically refers to the programming language as an entity
- "MicroPython language" refers to general language/wording used in the context of MicroPython, to describe things

If there's a strong desire to remove "the" in this particular instance then the heading would need to be changed to something like "MicroPython language reference", but I still think having "The" is more descriptive.

> There's a convention to capitalize each word in English headings, but we don't follow it

Yes, my preference is not to have capitals for all words.

pfalcon, 2017-07-02:
> then the heading would need to be changed

It's actually would be my next proposal - that chapter doesn't really describe "the language" in the same way the corresponding CPython chapter https://docs.python.org/3/reference/index.html does. Nor there's really "The MicroPython language", as we actually implement the Python language quite faithfully. So, my proposal to rename the chapter to "MicroPython language and implementation" to emphasize that it mostly describes features of a particular implementation. In this case, imho the article can be dropped. The word "language" is still there, mostly to still be somehow consistent with CPython doc structure and differentiate it from the library chapter.

dpgeorge, 2017-07-03:
> So, my proposal to rename the chapter to "MicroPython language and implementation" to emphasize that it mostly describes features of a particular implementation.

Ok, that sounds fine.

But in general the use of "the" would be on a case-by-case basis.  Eg the left-hand ToC for the pyboard (http://docs.micropython.org/en/latest/pyboard/) has headings such as "Quick reference for the pyboard" and I would want to keep "the" in this heading.

dpgeorge, 2020-01-06:
This was fixed in de78a9e317d3dd3654424ea70446dedbae9cce52 and aec88ddf0326187c567d4e4507149fb7c54ba91d

closed 2020-01-06 by dpgeorge