micropython/micropython #3193: ESP8266 how to support the development board?

rmokerone, 2017-07-03:
I would like to use ``esp8266 micropython`` to make a development board.I had some questions before I started.
- Should I push the adaptation code to the master branch?Like ``stmhal`` do.
- Where should I create a folder to fit my board. Like ``esp8266/boards/``?
- How about giving each pin a specific name?

And so on, if there is a clear specification, will be grateful.
[image]

torwag, 2017-07-04:
Best to ask those question in the forum as it is not really an issue.
https://forum.micropython.org

dpgeorge, 2017-07-05:
The esp8266 port doesn't have a clear concept of "board", instead it's a port to the SoC itself.  So your board should work just fine without any special additions to the code.

> How about giving each pin a specific name?

Pins are referenced by an integer, being the pin number defined by the esp8266.  There's no support for naming them (you can do it in Python if you like, eg `PIN_D1 = 4`).

closed 2017-07-05 by dpgeorge