micropython/micropython #3194: Cannot change GPIO value

capriele, 2017-07-03:
Hi,
I'd like to change the value of a GPIO pin (on NUCLEO_F429ZI) connected to a led. I've tried different method on different pins. In particular the following code is working
`pyb.Pin(pyb.Pin.cpu.B0,pyb.Pin.OUT).on()`
istead this one is NOT working
`pyb.Pin(pyb.Pin.cpu.B14,pyb.Pin.OUT).on()`

Is this problem related to the fact that the UART_REPL? For this board UART_REPL = PYB_UART_3.
Thanks to all!

dpgeorge, 2017-07-04:
af=12 means that the pin is configured for USB D-.  I'm not familiar with this particular NUCLEO board but perhaps it has 2 USB connectors, and it looks like the HS one is configured to have serial IO on it.

capriele, 2017-07-04:
Ok is there a way to change this configuration?

capriele, 2017-07-04:
I've fixed this behaviour in this way: 

- I've commented out `//#define USE_USB_HS` and `//#define USE_USB_HS_IN_FS`
- I've defined `#define USE_USB_FS`
inside `stm32f4xx_hal_conf.h`

closed 2017-07-04 by capriele

dpgeorge, 2017-07-04:
Ok, it looks like this dev board only has the USB connector on the FS peripheral, so I changed it's config in d5ec46ace4bdd0fa3b2ca45371f537259f07a6f2