micropython/micropython #3195: RFC ESP8266 instability

peterhinch, 2017-07-04:
In the course of intensive work on the ESP8266 I have encountered three issues which I thought worth recording. They are not deterministic and I've not found a way to provoke them with a test script.

The commonest is that it gets into a mode where it repeatedly issues `LmacRxBlk:1`. Only a power cycle clears it. This can occur for no obvious reason - connecting with rshell, running a script with a Python syntax error, issuing ctrl-C to a running program, or just idling at the REPL prompt. In the following sequence the script correctly raises `OSError`. The sequence starts *after* the REPL appears.

```
>>> import test1
Connected to wifi
Connected to broker
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test1.py", line 68, in <module>
  File "uasyncio/core.py", line 121, in run_forever
  File "uasyncio/core.py", line 85, in run_forever
  File "test1.py", line 54, in main
  File "mqtt_as.py", line 193, in connect
  File "mqtt_as.py", line 190, in connect
  File "mqtt_as.py", line 71, in _as_read
OSError: -1
>>> LmacRxBlk:1
LmacRxBlk:1
[ad infinitum]
```

The second is the issuing Python syntax errors without a line number. This can occur while a script is running - the script continues to run correctly, and the behaviour is not deterministic. It can even occur at the REPL:
```python
>>> TypeError: 'int' object is not callable
```

Another is this sequence, occurring while a program is running and leading to a reboot:
```
Fatal exception 28(LoadProhibitedCause):
epc1=0x40264c92, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000070, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 31228, room 16 
tail 12
chksum 0xd5
ho 0 tail 12 room 4
load 0x3ffe8000, len 1080, room 12 
tail 12
chksum 0xcb
ho 0 tail 12 room 4
load 0x3ffe8440, len 3248, room 12 
tail 4
chksum 0xbe
csum 0xbe
[REBOOTS HERE]
```

dpgeorge, 2017-07-05:
That's interesting but it's going to be hard to get to the bottom of these issues.  The real question is whether they are uPy related, 3rd party software related (either lwip or internal Espressif code), or hardware related.  The latter 2 issues seem like ones that could potentially be fixed, if only they were reproducible.

peterhinch, 2017-07-05:
These issues seem to be provoked by the development process rather than by the actual application which starts reliably on power up if I start it via main.py. More testing of this is required to build confidence. It runs indefinitely - I have run it for ~36 hours continuously, and several times for periods of many hours, without problems.

I appreciate that nondeterministic faults are very hard to fix. A 'solution' might be to find a workflow which appears to avoid the issues and document it.

Currently I cross-compile, start rshell, copy the .mpy to the board, start the REPL, and run the test program. On completion (via ctrl-C or an error) I return to Bash. Is it possible that state is retained or some resource consumed on each iteration? Two alternatives which come to mind are:
 1. Power cycle each time I return to Bash.
 2. Issue a soft reset on entering the REPL and again before dropping back to Bash.

I'd be glad to hear any suggestions. Meanwhile I'll experiment with the above methods.

dpgeorge, 2017-07-06:
> Is it possible that state is retained or some resource consumed on each iteration?

Yes.  Soft reset doesn't work well on the esp8266.  I'd suggest you do a hard reset of the device in between each development cycle.

peterhinch, 2017-07-14:
An update. Hard resets have removed all but the `LmacRxBlk:1` issue.

My application uses about 50% of the ESP8266 RAM as frozen bytecode. It uses networking and uasyncio and has run successfully for many hours until interrupted.

If I run it and, after five minutes, interrupt it with ctrl-C I get a REPL. After a minute or so the board spontaneously and repeatedly issues `LmacRxBlk:1`; only a hard reset clears it. The same thing occurs if the code is adapted to exit normally after 5 minutes. It appears to shut down normally until, after an interval, `LmacRxBlk:1` is repeatedly issued.

I have observed instances of `LmacRxBlk:1` at runtime, but these occur sporadically at times when the application is under load and don't affect its operation.

pfalcon, 2017-07-14:
It's very well known why `LmacRxBlk:1` occurs - please search the tickets and read the documentation.

peterhinch, 2017-07-15:
OK, I get it now. The issue occurs when the application is interrupted with ctrl-C or if another unhandled exception is raised. In this instance the socket is not closed. So, at the REPL, it falls over after some time when the socket receives data. Fixed by trapping exceptions and closing the socket.

Easy when you know what's going on. But this behaviour at the REPL is unfamiliar, confusing and counter intuitive. Especially as a hard reset is required to clear it.

peterhinch, 2017-07-21:
Trapping all exceptions and closing the open socket has fixed all these stability issues. I'm no longer finding it necessary routinely to hard-reset the board. Since we've documented this I guess it can be closed.

closed 2017-07-21 by peterhinch

cefn, 2017-12-28:
Thought it worth noting [this recent comment](https://github.com/esp8266/Arduino/issues/928#issuecomment-338065755) from the esp8266/Arduino project.

> LmacRxBlk:1 can also be solved by lwip update #3362. It was in fact the trigger.

Although understanding the relationship between all these projects and comprehension of the LmacRxBlk overrun is above my pay grade, this comment looked superficially promising that basing work on a newer version of Lwip, (now that Expressif is sharing their Lwip source), might resolve the underlying issue.

cefn, 2017-12-28:
OK, I found that this is already under discussion [here](https://github.com/pfalcon/esp-open-sdk/issues/263#issuecomment-303092791)