micropython/micropython #3196: mpy build error

ashareye, 2017-07-04:
I got error messages when building mpy-cross with macos 10.12.5 & clang 802.0.42

Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
Traceback (most recent call last):
  File "../py/makeversionhdr.py", line 107, in <module>
    make_version_header(sys.argv[1])
  File "../py/makeversionhdr.py", line 77, in make_version_header
    git_tag, git_hash, ver = info
TypeError: 'NoneType' object is not iterable
make: *** [build/genhdr/mpversion.h] Error 1

ashareye, 2017-07-04:
I tried using "make V = 1", but still the same result

dpgeorge, 2017-07-04:
Thanks for the report, should be fixed by b51919f5b757ee6409f97b93c0f0a8bec84f55ad

closed 2017-07-04 by dpgeorge

ashareye, 2017-07-04:
ths for the quick reply & fixing, its ok now!