micropython/micropython #3197: compat 2/3 : incorrect support of   hash()  on generator function/object

pmp-p, 2017-07-05:
MicroPython v1.9-11-g00e4f05-dirty on 2017-06-09; linux version
>>> def generate():   yield 1
...     
>>> g = generate()
>>> g
<generator object 'generate' at db737200>
>>> hash(g)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported type for __hash__: 'generator'

dpgeorge, 2017-07-06:
See also #2907 

What is your particular use-case for hashing generator functions?

pmp-p, 2017-07-06:
i was attempting to use bluelet coroutining as-is, which i use on python2/3 and python 2.7 emscripten port ( no theading allowed ).

dpgeorge, 2017-07-07:
Ok, it turns out this can be implemented at no cost (to code or RAM usage), see f69ab79ec8347c1f5ff0c6f31947ec06073fbd52

@pmp-p BTW did you know about uasyncio support in uPy?

closed 2017-07-07 by dpgeorge

pmp-p, 2017-07-07:
Thank you ! i had lot of - working - python2  code to port that rely on bluelet ( and i like asyncio syntax as much as systemd )