micropython/micropython #3198: ssl_handshake_status: -40 error

kannan95, 2017-07-07:
When i tried to send a POST request to this URL [https://www.terasyshub.io/api/v1/data](url) I'm getting ssl_handshake_status: -40 error. I have tried flashing all the latest builds including today's build. But still getting the same error. I have ran the test script from  #3150  with including my URL, the output I'm getting was this 

> google.com ok
www.google.com ok
api.telegram.org ok
www.terasyshub.io/api/v1/data OSError(-2,)
w9rybpfril.execute-api.ap-southeast-2.amazonaws.com ok

 The url I included only throwing error.

dpgeorge, 2017-07-08:
According to https://www.ssllabs.com/ssltest/ the server www.terasyshub.io only allows DHE and ECDHE ciphers and axtls doesn't support either of these, so the unix and esp8266 ports won't be able to connect to this server.

If you have access to the terasyshub.io server then you can enabled other ciphers to get it to work.  An alternative solution is to convert unix and esp8266 to use mbedtls.

kannan95, 2017-07-08:
What ciphers can I use instead of DHE and ECDHE? Can you provide the ciphers with are supporting in axtls and esp8266. I should able to send POST request to the server, the cipher suite should work. Can i use the cipher suite suggested in [https://github.com/igrr/axtls-8266](url) ?

>  An alternative solution is to convert unix and esp8266 to use mbedtls.
I have no clue what you said here. Can you guide me how to convert unix and esp8266 to use mbedtls?

pfalcon, 2017-07-08:
> What ...

You should refer to axTLS site/documentation to find answers to such questions. http://axtls.sourceforge.net/

KaikyuLotus, 2020-09-09:
Is there a solution for this issue?
I can't get api.telegram.org to work with urequests

I'm using esp8266 2020-09-06 unstable-v1.13-4
The connection to other HTTPS urls works

```
>>> urequests.get("https://api.telegram.org")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/urequests.py", line 112, in get
  File "/lib/urequests.py", line 60, in request
OSError: -40
```

dpgeorge, 2020-09-11:
> Is there a solution for this issue?

The only way to use DHE is to switch esp8266 to use mbedtls.  The `api.telegram.org` site, with TLS 1.2, needs ECDHE-RSA-AES128-GCM-SHA256 and that is supported by mbedtls.

I tried switching esp8266 to mbedtls and it does work, the esp8266 is able to connect to `api.telegram.org`.  But because this cipher is very complex it takes a long time for the esp8266 to process the encryption and it loses wifi while running the algorithm.  To fix that I inserted calls to keep the wifi stack running during mbedtls but it's a bit of a hack.

In summary: it is possible to get the esp8266 to use mbedtls and connect to `api.telegram.org`, but it's really pushing the limits of this device.

dpgeorge, 2020-09-11:
> it is possible to get the esp8266 to use mbedtls and connect to `api.telegram.org`,

See #6433 for the patch that allows this.

laurento, 2021-07-18:
Many thanks @dpgeorge I just tried your patch against the latest master branch, and it applies with only a minor issue.

Unfortunately, it doesn't seem to work here.
```
import urequests as requests

res = requests.get(url='https://www.google.com')  # no exception

res = requests.get(url='https://api.telegram.org')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "urequests.py", line 116, in get
  File "urequests.py", line 62, in request
OSError: (-30592, 'MBEDTLS_ERR_SSL_FATAL_ALERT_MESSAGE')
```
Any ideas?

jonnor, 2024-09-02:
This is a limitation of axtls which unfortunately we cannot fix. People can try the mbedtls, which can be switched with a configuration - but available memory might be very tight.
The best would be to migrate away from ESP8266 for applications that need modern SSL.

closed 2024-09-02 by jonnor