micropython/micropython #3199: Documentation outdated for latest machine.Pin API

cefn, 2017-07-07:
The documentation for Pin says there is a mode method on machine.Pin...

> Pin.mode([mode])
> Get or set the pin mode. See the constructor documentation for details of the mode argument.

https://docs.micropython.org/en/latest/esp8266/library/machine.Pin.html

However from an interactive session with the latest image I get...

```
MicroPython v1.9.1-8-g7213e78d on 2017-06-12; ESP module with ESP8266
Type "help()" for more information.
>>> from machine import Pin
>>> pin = Pin(0)
>>> pin.mode()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Pin' object has no attribute 'mode'
```

As a workaround I have been aiming to use init() instead, like e.g.
 
`pin.init(mode=Pin.IN)`

dpgeorge, 2017-07-10:
The reason pin.mode() and pin.pull() are there is because the cc3200 port implement them, and it was the first to have the machine module.  These methods are not available on all other ports and a decision needs to be made if they stay or go.

> As a workaround I have been aiming to use init() instead, like e.g.
> 
> pin.init(mode=Pin.IN)

At the moment this won't work as expected (on esp8266) because it'll also reset the pull resistors to no pull.  So if the mode/pull methods are removed it would be good to enhance the init() method so it can change individual settings.

cefn, 2017-07-10:
Thanks, @dpgeorge. 

Do you mean you need to explicitly call `pin.init(mode=Pin.IN, pull=Pin.PULL_UP) ` if you wanted to continue with a specific pull configuration, or are you implying that wouldn't work either?

As it turns out I don't have any pull configuration and wasn't expecting any to be set by either Pin(0) or my call `pin.init(mode=Pin.IN)`. I was expecting that my call would implicitly assert `pull=-1` as per the `init()` documentation, indicating no pull and therefore make no change to pull.

From your comment, a better-worded issue could be to seek one of the following...

* the documentation should template-filter the docstrings to prevent them appearing in esp8266
* the documentation should indicate mode() and pull() are restricted to a specific cc3200 port
* all implementation and reference to pull() or mode() should be dropped

dpgeorge, 2017-07-10:
> Do you mean you need to explicitly call pin.init(mode=Pin.IN, pull=Pin.PULL_UP) if you wanted to continue with a specific pull configuration, or are you implying that wouldn't work either?

Yes, you'll need to specify the pull mode you're using in the call to init.  If you're not using any pull then no need to specify anything.  Basically, pass the same parameters here that you did when creating the Pin object for the first time.

pfalcon, 2017-07-10:
@cefn 

> The documentation for Pin says there is a mode method on machine.Pin...

The first and foremost thing that doc says is: http://docs.micropython.org/en/latest/pyboard/library/index.html

> Any particular MicroPython variant or port may miss any feature/function described in this general documentation, due to resource constraints.

pfalcon, 2017-07-10:
> These methods are not available on all other ports and a decision needs to be made if they stay or go.

I'm +1 for removing them from the docs.

> it would be good to enhance the init() method so it can change individual settings.

I think .config() method of network module (NIC objects) is a good design success. I believe in Hardware API ticket there was a proposal to rename .init() to .config(). I'm now +1 on that, and following the same API as in network module (have defined set of properties, but allow extensions; allow to both set and query properties).

dpgeorge, 2017-07-11:
> I believe in Hardware API ticket there was a proposal to rename .init() to .config().

I'm quite -1 on this, it would be such a breaking change, to rename it.  Also there's deinit() which would no longer be paired with init().

And how to distinguish between those parameters that can be passed to the constructor vs those that are passed to a hypothetical config() method?

pfalcon, 2017-07-11:
> I'm quite -1 on this, it would be such a breaking change, to rename it.

We can use previous experience, and keep init() as a deprecated alias for config "until v2.0" or something.

> Also there's deinit() which would no longer be paired with init().

There're always drawbacks and compromises for the kind of changes we're doing now, all low-hanging "definitely positive" things were already done.

> And how to distinguish between those parameters that can be passed to the constructor vs those that are passed to a hypothetical config() method?

There's no distinction. The reason methods likes mode()/pull() etc. exist is because they allow to query corresponding property. And that would be the only extension to init-renamed-as-config - to let it be used to query params. That's also the reason behind renaming.

dpgeorge, 2017-07-11:
> We can use previous experience, and keep init() as a deprecated alias for config "until v2.0" or something.

But the semantics are different: currently uart.init(9600) will change the baudrate, but if it's config() then that will raise an exception because 9600 is not a string.

pfalcon, 2017-07-11:
Well, that's rough idea anyway, which I considered only for Pin, and for which I wanted to submit RFC, but didn't, because didn't have time to prepare well (which includes reviewing current wider .init() usage). Cases like above of course can be handled with some level of hacks (to be removed together with .init() later if the idea would go ahead).

dpgeorge, 2017-07-11:
Why not just make minimal changes and extend the spec of init() to allow to set individual items?  That's already what some ports try to do.

pfalcon, 2017-07-11:
Well, the discussion started with a point that methods like pull() exist because they allow to query the state, and how to generalize that, using existing API patterns. Not that I personally consider "querying" usecase important, my usual argument would be that it's not possible in the general case. But leaving room for it may be a good idea, and reusing existing patterns (.config()) seems like good idea, including from the point of "not possible in the general case" (so, adding a dedicated method for that is worse imho).

mattytrentini, 2022-03-07:
This is an old ticket and the docs appear to have been updated; [Pin](https://docs.micropython.org/en/latest/library/machine.Pin.html) seems clearly documented now. Can we close this ticket?

closed 2022-03-07 by dpgeorge