micropython/micropython #3202: Build failure on osx, py/compile.c

hoihu, 2017-07-08:
91a385db98711c2a9115ad661cb4d637450d7ee2

introduces reproducably a build fault on my system (running on OS x, gcc 4.8) for the mpy-cross and stmhal target (pyboard):

```
GEN build/genhdr/qstr.i.last
GEN build/genhdr/qstr.split
GEN build/genhdr/qstrdefs.collected.h
QSTR not updated
CC ../py/compile.c
../py/compile.c:2164:38: error: implicit conversion from enumeration type 'mp_unary_op_t' to different enumeration type
      'mp_binary_op_t' [-Werror,-Wenum-conversion]
        case MP_TOKEN_OP_PLUS:  op = MP_UNARY_OP_POSITIVE; break;
                                   ~ ^~~~~~~~~~~~~~~~~~~~
../py/compile.c:2165:38: error: implicit conversion from enumeration type 'mp_unary_op_t' to different enumeration type
      'mp_binary_op_t' [-Werror,-Wenum-conversion]
        case MP_TOKEN_OP_MINUS: op = MP_UNARY_OP_NEGATIVE; break;
                                   ~ ^~~~~~~~~~~~~~~~~~~~
../py/compile.c:2168:18: error: implicit conversion from enumeration type 'mp_unary_op_t' to different enumeration type
      'mp_binary_op_t' [-Werror,-Wenum-conversion]
            op = MP_UNARY_OP_INVERT;
               ~ ^~~~~~~~~~~~~~~~~~
../py/compile.c:2171:24: error: implicit conversion from enumeration type 'mp_binary_op_t' to different enumeration type
      'mp_unary_op_t' [-Werror,-Wenum-conversion]
    EMIT_ARG(unary_op, op);
    ~~~~~~~~~~~~~~~~~~~^~~
../py/compile.c:75:58: note: expanded from macro 'EMIT_ARG'
#define EMIT_ARG(fun, ...) (mp_emit_bc_##fun(comp->emit, __VA_ARGS__))
                            ~~~~~~~~~~~~~~~~             ^~~~~~~~~~~
4 errors generated.
make: *** [build/py/compile.o] Error 1

```

Has anyone observed this as well or did I hit some mac-specific build phanomena?

hoihu, 2017-07-09:
https://github.com/micropython/micropython/blob/master/py/compile.c#L2161

looks suspicous to me. Shouldn't it read
`mp_unary_op_t op;` ?

Changing it to that compiles fine afterwards.

dpgeorge, 2017-07-09:
Thanks for the report.  I couldn't reproduce this warning but fixed it as suggested above in 0291a624cbb0fc12633291b5b53285f7425dd3dc

closed 2017-07-09 by dpgeorge

hoihu, 2017-07-09:
Thanks. 

As a sidenote: It is strange that the error pops up on my machine, but not in travis. I retested this with the same gcc version 4.9.3 as used by travis:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
```
Maybe there are differences between gcc versions from the mac to the linux build