micropython/micropython #3203: arm-linux-androideabi-gcc: not found on RasPi 3

mdupuy, 2017-07-08:
Building for Debian (raspbian) via the standard Getting Started instructions: 
https://github.com/micropython/micropython/wiki/Getting-Started and running:
`make axtls`

Yields en error indicating it was looking for the Android NDK gcc, not the stock, 4.9 ARMv7 hard float compiler.

```
make[1]: Entering directory '/home/mdupuy/dev/micropython/micropython/lib/axtls'
make -C crypto
make[2]: Entering directory '/home/mdupuy/dev/micropython/micropython/lib/axtls/crypto'
/bin/sh: 1: arm-linux-androideabi-gcc: not found
make[2]: *** No rule to make target '.depend', needed by 'all'.  Stop.
```

My particulars:
```
: uname -a
Linux rapi 4.9.36-v7+ #1015 SMP Thu Jul 6 16:14:20 BST 2017 armv7l GNU/Linux

: gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Raspbian 4.9.2-10) 

```

dpgeorge, 2017-07-10:
We need more information to help here.  Can you please post the entire output from the command `make V=1 axtls` (from a fresh repo), and also the output of `env` (from which you may need to remove some sensitive info).

mdupuy, 2017-07-10:
Ah, you caught me. I'm an idiot. I had an old "CROSS_COMPILE" environment setting in my shell config. Clearing it out used the native compiler. How embarrassing.

closed 2017-07-10 by mdupuy