micropython/micropython #3204: Unix: CPython incompatibility assigning negative values to bytearrays.

peterhinch, 2017-07-09:
This is arguably minor, but I thought it worth flagging up. It also applies to bytes objects.
```python
MicroPython v1.8.7-893-g46b849a on 2017-06-22; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> a = bytearray([0, 1, -1])
>>> a[2]
255
>>> 
```
Under CPython 3.4.3:
```python
>>> a = bytearray([0, 1, -1])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: byte must be in range(0, 256)
```

pfalcon, 2017-07-09:
If there's a desire to fix this (which shouldn't arise without real-world usecases being presented), the check should be wrapped in MICROPY_FULL_CHECKS, like already done for bytes().

jerryneedell, 2018-11-28:
FYI - in a related issue I ran into this difference in behavior between CircuitPython (and presumably Micropython) and Python 3.6 under Linux

When a variable stored as a byte overlflows, on CircuiPython, it just "wraps" but on Python 3.6 it throws an error.

The CircuitPython behavior is more like "C" if the variable is declared as a  uint8_t.

CitcuitPython 
```
Adafruit CircuitPython 4.0.0-alpha.3-25-gf6c73b35f on 2018-11-27; Adafruit ItsyBitsy M4 Express with samd51g19
= bytearray([0x80,0x70,0x60,0x50])
>>> x
>>> x[0] = x[0]<<1
>>> x[0]
0
>>> 
```

Python 3.6 on Linux
```
Python 3.6.7 (default, Oct 22 2018, 11:32:17) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
>>> x = bytearray([0x80,0x70,0x60,0x50])
>>> x[0] = x[0] <<1
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: byte must be in range(0, 256)
>>> 
```

jonnor, 2024-09-19:
Still looks to happen as of MicroPython v1.24.0-preview.276.g1897fe6227

BillKujawa, 2025-04-04:
What have I missed? It seems the Docs indicate that this should work.

'''
import struct
"""
    Negative byte to int not converting to negative number
    Micropython (Raspberry Pi Pico W)
    RPI_PICO_W-20241129-v1.24.1.uf2
    
    Docs:
    to_bytes method doesn’t implement signed parameter.
    Cause: The signed keyword-only parameter is not implemented for int.to_bytes().
    When the integer is negative, MicroPython behaves the same as CPython int.to_bytes(..., signed=True)
    When the integer is non-negative, MicroPython behaves the same as CPython int.to_bytes(..., signed=False).
"""
pkt = bytearray(2)
fmt = '>h'
num = -2

print(f"{num=}")
sz = struct.calcsize('>h')
print(f"{sz=}")

struct.pack_into(fmt, pkt, 0, num)
print(f"{pkt=}")

pkt_view = memoryview(pkt)
temp = int.from_bytes(pkt_view[0:2], "big")
print(f"{temp=}")

"""
>>> %Run -c $EDITOR_CONTENT (thonny)
num=-2
sz=2
pkt=bytearray(b'\xff\xfe')
temp=65534
"""
'''

dlech, 2025-04-04:
> What have I missed?

The documentation you quoted is for `to_bytes()` but your code uses `from_bytes()`?

robert-hh, 2025-04-04:
See also https://github.com/micropython/micropython/issues/15399

robert-hh, 2025-04-04:
Why not using struct.upack() to get the data from the buffer? That will handle the sign properly.

BillKujawa, 2025-04-04:
Sounds good. I will recode.