micropython/micropython #3205: Super doesn't find parent methods

dvrhax, 2017-07-09:
Discovered this issue while porting over PyMySQL to micropython both super() calling methods shown produce the same results 

```python
class a():
    def doSomething(self):
        print('Something done.')

class aMixin():
    def doSomething(self):
        #super(aMixin, self).doSomething()
        super().doSomething()
        print('Something Else.')

class b(aMixin, a):
    """Super Test Case"""

print('Class a:')
aa = a()
aa.doSomething()

print('Class b:')
bb = b()
bb.doSomething()
```

CPython prints:
>Class a:
>Something done.
>Class b:
>Something done.
>Something Else.

However MicroPython produces the following error:
>Class a:
>Something done.
>Class b:
>Traceback (most recent call last):
>  File "<stdin>", line 1, in <module>
>  File "superTest.py", line 19, in <module>
>  File "superTest.py", line 7, in doSomething
>AttributeError: 'super' object has no attribute 'doSomething'

pfalcon, 2017-07-09:
See http://docs.micropython.org/en/latest/pyboard/genrst/core_language.html#classes , 2nd case. This is won't fix. Workaround: avoid such complex, potentially confusing for a human patterns in MicroPython apps.

dvrhax, 2017-07-09:
I would tend to agree.  It took me a long time to figure out what the code in PyMySQL was doing when I came across this error.  I ended up resolving it by rewriting the code with a more simple inheritance scheme.  However I will caveat that with I had to write a lot of redundant code to get it to work and on a storage limited device having this capability might be an advantage.

pfalcon, 2017-07-09:
There's no need to have redundant code. You can abstract it away to any method or function to reuse. The key is not getting carried away with trying to reuse the same method name for too many different things (and then drown in trying to "inherit" or "override" them.)

pfalcon, 2017-08-23:
As mentioned above, this is a documented difference with "big" Python. Closing.

closed 2017-08-23 by pfalcon