micropython/micropython #3206: stmhal: ESPRUINO_PICO build failed

dmazzella, 2017-07-11:
with master configuration ESPRUINO PICO build fail with error: region `FLASH_TEXT' overflowed by 7820 bytes

i have removed all files from "modules" and added an empty __empty__.py

dmazzella, 2017-07-11:
can be this a correct configuration for stm32f401xd.ld:

```
    FLASH (rx)      : ORIGIN = 0x08000000, LENGTH = 384K /* entire flash, 384 KiB */
    FLASH_ISR (rx)  : ORIGIN = 0x08000000, LENGTH = 16K /* sector 0, 16 KiB */
    FLASH_FS (rx)   : ORIGIN = 0x08004000, LENGTH = 48K /* sectors 1,2,3 are 16K.  */
    FLASH_TEXT (rx) : ORIGIN = 0x08020000, LENGTH = 320K /* sectors 4,5,6. 4 is 64K, 5 and 6 are 128KiB = 320K KiB */
    RAM (xrw)       : ORIGIN = 0x20000000, LENGTH = 96K /* 96 KiB */
```

pfalcon, 2017-07-11:
@dpgeorge : Of course tickets like this has nothing to do with discussion in https://github.com/micropython/micropython/pull/3200 and other similar ones.

dpgeorge, 2017-07-11:
> Of course tickets like this has nothing to do with discussion in #3200 and other similar ones.

Yes I know.  The issue here is that the HAL part of stmhal can be simplified and reduced by quite a lot, and also the frozen module configuration should really be configurable per board.

dpgeorge, 2017-07-11:
> can be this a correct configuration for stm32f401xd.ld:

@dmazzella no, that config will break the filesystem since it would only leave 3*16k=48k for it, which is not enough.

pfalcon, 2017-07-11:
@dpgeorge That's specifics. The real cause is that uPy constantly growth, and without controlling growth, it will just explode, and there will be more and more reports like that (just like we keep getting reports for pseudo-missing functionality which was rejected long ago).

dpgeorge, 2017-07-11:
> The real cause is that uPy constantly growth

Please take note of the many commits I did over the past year reducing the code size while retaining the same functionality.  Note that v1.9 cf v1.8.7 dropped by over 1k code size in bare-arm, minimal and unix.  So in fact uPy shrinks, and can then afford to include things to make it even more like CPython.

pfalcon, 2017-07-11:
Yes, for example, to include sys.getsizeof(), which is important memory usage debugging measure, and which doesn't have any other alternative, which I wanted to implement for a while and will post initial patch for. Not duplicates of the more efficient functionality that already exists. (This gets offtopic in this ticket.)

dmazzella, 2017-07-11:
> @dmazzella no, that config will break the filesystem since it would only leave 3*16k=48k for it, which is not enough.

i have tried this configuration and it work, 48K is little but sufficient fs

dpgeorge, 2017-07-12:
> i have tried this configuration and it work, 48K is little but sufficient fs

Ok, good to know.

In f1d260d878105bfbf25c0bb68da6190e35fc106a the Pico build was modified to remove the default frozen modules, native/viper/asm_thumb emitters and complex numbers.  This retains the 112k filesystem and gives about 14k of flash to spare.  Anyone needing specific features like asm_thumb on the Pico will need to tune the build to their needs.

closed 2017-07-12 by dpgeorge