micropython/micropython #3208: close method of SSL-wrapped sockets is not idempotent

alex-robbins, 2017-07-12:
On MicroPython (as well as CPython), file objects and sockets can both have their `close` methods called multiple times (calls after the first have no effect; the operation is idempotent). On MicroPython (but not CPython), this is not the case for the objects returned by `ussl.wrap_socket`. With these objects, a second call to `close` causes a crash (a segfault on Unix and a reset on the ESP8266).

A closed SSL socket still has normal-ish (non-crashing) behavior with other functions, so I guess nothing's being garbage collected or deallocated or anything. 

I think the behavior is undesirable because, besides being unexpected and incompatible with CPython, it also makes error handling for user-defined classes tricky in the event that the socket in question may or may not have already been closed. There is also no way to test a socket to see if it has been closed.

alex-robbins, 2017-07-13:
I should have mentioned that I'm using axTLS for both unix and esp8266 (whatever the newest one is in master).

Backtrace: 
(Does anyone actually find this helpful, or is it just more noise? I mainly just looked at it for my own edification; I'm guessing you guys already know the relevant code path here.)
```
#0  AES_cbc_encrypt (ctx=0x0, msg=0x555555801595 "\006|\304\302_QO(\313\033y\277\070ܝm\001", 
    out=0x555555801595 "\006|\304\302_QO(\313\033y\277\070ܝm\001", length=48) at aes.c:274
#1  0x00005555555cb34f in send_packet (ssl=0x555555801550, protocol=protocol@entry=21 '\025', 
    in=in@entry=0x7fffffffc5fe "\001", length=length@entry=2) at tls1.c:1187
#2  0x00005555555cb677 in send_alert (ssl=ssl@entry=0x555555801550, error_code=error_code@entry=0) at tls1.c:1658
#3  0x00005555555cbd0b in ssl_free (ssl=0x555555801550) at tls1.c:256
#4  0x00005555555ba5b6 in socket_close (self_in=0x7ffff6f91520) at ../py/../extmod/modussl_axtls.c:154
```

Would an appropriate solution be to set one of the pointer members of `mp_obj_ssl_socket_t` to NULL on the first close, and test for NULL to make subsequent calls no-ops? That only sounds like a handful of bytes...

dpgeorge, 2017-07-13:
Good find.  Yes this should be fixed.  The hard part will be finding a minimal way to do it so that read/write on a closed socket won't crash.

pfalcon, 2017-07-19:
Fixed in 4368ae31424f93f3272209ea61847f2406dd23ad

closed 2017-07-19 by pfalcon