micropython/micropython #3209: NodeMCU ARNICA

adrennes, 2017-07-13:
I bought a NodeMCU from AMAZON equiped with ESP8266. I tried to use https://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html 
with no success.
At the end, here is issue: NodeMCU bought from AMAZON designed by AI-Thinker (http://www.ai-thinker.com) comes with ESP8266-12F and not 12E as in the quick ref. There are pinout changes, i.e. the I2C: SDA is D4 and SCL is D5 of ESP-12E.
Correct instanciation must refer to ESP-12F GPIO number like this :
from machine import I2C, Pin
import time
i2c = I2C(scl=Pin(14), sda=Pin(2), freq=50000)
[NodeMCU_ESP-12F.pdf](https://github.com/micropython/micropython/files/1146034/NodeMCU_ESP-12F.pdf)

This works well.
Pull-up resistors are not necessary with this ESP-12F.

Last point: how to identify ESP-12F ? look at the antenna shape: it is different (see my pdf).
It should be great and usefull to update the quick reference with the above info.
I love micropython.
Regards,
Alain DUBREUIL

robert-hh, 2017-07-13:
There are other boards too, like the WEMOS D1 mini, where the numbering on the board does not match the GPIO port numbers of the ESP8266. The examples are referring to the ESP8266 GPIO numbers. The mapping between board number and ESP8266 GPIO number has then to be made by the user. For the reference board, the Huzzah ESP8266 feather, the labeling is consistent.

dpgeorge, 2017-07-14:
It looks like ESP-12E and ESP-12F have the same pinout, so if there's anything that's different it's with the NodeMCU board.  The esp8266 port doesn't have a concept of "board" but rather supports the generic esp8266 chip.  As @robert-hh says you need to work out the pin mappings for your particular board to get it working.

closed 2017-07-14 by dpgeorge