micropython/micropython #3210: Modification to SPI, I2C in mpconfigport.py in ESP8266 makes code to not build

skbobade, 2017-07-14:
What I did: Changed lines of code in mpconfigport.py to disable I2C and SPI like so - 
`#define MICROPY_PY_MACHINE_I2C (0)`

`#define MICROPY_PY_MACHINE_SPI (0)`

Tried with `#define MICROPY_PY_MACHINE_SPI_MAKE_NEW machine_hspi_make_new` present as is and commented.

In any combination of these code changes, I find that subsequent build using `make -B` fails.

See attached file for output of the `make -B` command.

dpgeorge, 2017-07-17:
The esp8266 port is not designed to disable arbitrary config options in mpconfigport.h because there are quite a few dependencies amongst the components.  In this case, to disable I2C/SPI, you'll also need to edit esp8266/modmachine.c and delete lines 259/260 (those with links to machine_i2c_type/machine_hspi_type).

dpgeorge, 2017-07-17:
> The esp8266 port is not designed to disable arbitrary config options in mpconfigport.h

Well, this statement is in slight contradiction with the 512k port which *is* intended to be tuned by an advanced user, to fit the required features in 512k.  SPI/I2C could be made easily configurable by adding more #if guards in the code.

skbobade, 2017-07-19:
Thanks for reopening the issue. Hope it will get addressed soon enough. FYI - There are other definitions in esp8266 port, which will break the build when disabled, example, `MICROPY_HELPER_REPL`

So maybe only a few configurations are supposed to be configurable, and there are others which shouldn't be changed? This may need to be updated in the documentation.

dpgeorge, 2018-05-17:
This is fixed in 58331e3c28e41ad7e3c3dd7dabf6d508a8bfe1cb where I2C and SPI can now be individually disabled.

closed 2018-05-17 by dpgeorge